shivasiddharth/GassistPi #182: unable to install on Rpi B+

SoundGuy, 2018-02-04:
I ran  sudo  /home/pi/GassistPi/scripts/gassist-installer-pi-zero.sh
I got this error:
Collecting protobuf>=3.0.0 (from gmusicapi>=11.0.0->-r /home/pi/GassistPi/Requir                                                                    ements/GassistPi-pip-requirements.txt (line 6))
Exception:
Traceback (most recent call last):
  File "/home/pi/env/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/url                                                                    lib3/connectionpool.py", line 594, in urlopen
    chunked=chunked)
  File "/home/pi/env/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/url                                                                    lib3/connectionpool.py", line 391, in _make_request
    six.raise_from(e, None)
  File "<string>", line 2, in raise_from
  File "/home/pi/env/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/url                                                                    lib3/connectionpool.py", line 387, in _make_request
    httplib_response = conn.getresponse()
  File "/usr/lib/python3.5/http/client.py", line 1198, in getresponse
    response.begin()
  File "/usr/lib/python3.5/http/client.py", line 297, in begin
    version, status, reason = self._read_status()
  File "/usr/lib/python3.5/http/client.py", line 266, in _read_status
    raise RemoteDisconnected("Remote end closed connection without"
http.client.RemoteDisconnected: Remote end closed connection without response

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/pip/basecommand.py", line 215,                                                                     in main
    status = self.run(options, args)
  File "/home/pi/env/lib/python3.5/site-packages/pip/commands/install.py", line                                                                     353, in run
    wb.build(autobuilding=True)
  File "/home/pi/env/lib/python3.5/site-packages/pip/wheel.py", line 749, in bui                                                                    ld
    self.requirement_set.prepare_files(self.finder)
  File "/home/pi/env/lib/python3.5/site-packages/pip/req/req_set.py", line 380,                                                                     in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/home/pi/env/lib/python3.5/site-packages/pip/req/req_set.py", line 554,                                                                     in _prepare_file
    require_hashes
  File "/home/pi/env/lib/python3.5/site-packages/pip/req/req_install.py", line 2                                                                    78, in populate_link
    self.link = finder.find_requirement(self, upgrade)
  File "/home/pi/env/lib/python3.5/site-packages/pip/index.py", line 465, in fin                                                                    d_requirement
    all_candidates = self.find_all_candidates(req.name)
  File "/home/pi/env/lib/python3.5/site-packages/pip/index.py", line 423, in fin                                                                    d_all_candidates
    for page in self._get_pages(url_locations, project_name):
  File "/home/pi/env/lib/python3.5/site-packages/pip/index.py", line 568, in _ge                                                                    t_pages
    page = self._get_page(location)
  File "/home/pi/env/lib/python3.5/site-packages/pip/index.py", line 683, in _ge                                                                    t_page
    return HTMLPage.get_page(link, session=self.session)
  File "/home/pi/env/lib/python3.5/site-packages/pip/index.py", line 792, in get                                                                    _page
    "Cache-Control": "max-age=600",
  File "/home/pi/env/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/re                                                                    quests/sessions.py", line 501, in get
    return self.request('GET', url, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/pip/download.py", line 386, in                                                                     request
    return super(PipSession, self).request(method, url, *args, **kwargs)
  File "/home/pi/env/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/re                                                                    quests/sessions.py", line 488, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/pi/env/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/re                                                                    quests/sessions.py", line 609, in send
    r = adapter.send(request, **kwargs)
  File "/home/pi/env/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.wh                                                                    l/cachecontrol/adapter.py", line 47, in send
    resp = super(CacheControlAdapter, self).send(request, **kw)
  File "/home/pi/env/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/re                                                                    quests/adapters.py", line 423, in send
    timeout=timeout
  File "/home/pi/env/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/url                                                                    lib3/connectionpool.py", line 643, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "/home/pi/env/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/url                                                                    lib3/util/retry.py", line 315, in increment
    total -= 1
TypeError: unsupported operand type(s) for -=: 'Retry' and 'int'

shivasiddharth, 2018-02-04:
Its a network error. try again after sometime.

SoundGuy, 2018-02-04:
ok. will try and keep you posted

SoundGuy, 2018-02-04:
new error : 

Collecting pycryptodomex>=3.0 (from gpsoauth>=0.2.0->gmusicapi>=11.0.0->-r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 6))
  Using cached pycryptodomex-3.4.11.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-f_o63qcu/pycryptodomex/setup.py", line 107, in <module>
        for line in open(os.path.join("lib", "Crypto", "__init__.py")):
    FileNotFoundError: [Errno 2] No such file or directory: 'lib/Crypto/__init__.py'

    -----------------------------------```
-----
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-f_o63qcu/pycryptodomex/

```

is it the same thing ? network error?

shivasiddharth, 2018-02-04:
open /tmp directory. Delete folders beginning with the name pip and then re-run the installer.

SoundGuy, 2018-02-05:
ls -la /tmp
total 48
drwxrwxrwt 11 root root 4096 Feb  5 06:17 .
drwxr-xr-x 21 root root 4096 Nov 29 04:57 ..
drwx------  2 pi   pi   4096 Feb  4 18:11 dhcpcd-(null)
drwxrwxrwt  2 root root 4096 Feb  4 18:09 .font-unix
drwxrwxrwt  2 root root 4096 Feb  4 18:09 .ICE-unix
drwx------  2 pi   pi   4096 Feb  4 18:10 ssh-kHfijrE6AfxQ
drwx------  2 pi   pi   4096 Feb  4 18:10 ssh-sHVj3DFRm1fj
drwx------  3 root root 4096 Feb  4 18:09 systemd-private-c15772696e364fb1a9aa7e0851a5d224-systemd-timesyncd.service-38pifa
drwxrwxrwt  2 root root 4096 Feb  4 18:09 .Test-unix
-r--r--r--  1 root root   11 Feb  4 18:09 .X0-lock
drwxrwxrwt  2 root root 4096 Feb  4 18:09 .X11-unix
drwxrwxrwt  2 root root 4096 Feb  4 18:09 .XIM-unix

SoundGuy, 2018-02-05:
ok. i just re-ran the install and now it reached the end and  i got this:

INFO:root:Transcript of user request: "volume  up".
INFO:root:Playing assistant response.
INFO:root:End of audio request detected
INFO:root:Transcript of user request: "volume up".
INFO:root:Playing assistant response.
INFO:root:Setting volume to 60%
INFO:root:Finished playing assistant response.
Traceback (most recent call last):
  File "/home/pi/env/bin/googlesamples-assistant-pushtotalk", line 11, in <module>
    sys.exit(main())
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/pushtotalk.py", line 424, in main
    continue_conversation = assistant.assist()
  File "/home/pi/env/lib/python3.5/site-packages/tenacity/__init__.py", line 214, in wrapped_f
    return self.call(f, *args, **kw)
  File "/home/pi/env/lib/python3.5/site-packages/tenacity/__init__.py", line 295, in call
    start_time=start_time)
  File "/home/pi/env/lib/python3.5/site-packages/tenacity/__init__.py", line 252, in iter
    return fut.result()
  File "/usr/lib/python3.5/concurrent/futures/_base.py", line 398, in result
    return self.__get_result()
  File "/usr/lib/python3.5/concurrent/futures/_base.py", line 357, in __get_result
    raise self._exception
  File "/home/pi/env/lib/python3.5/site-packages/tenacity/__init__.py", line 298, in call
    result = fn(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/pushtotalk.py", line 169, in assist
    self.conversation_stream.stop_playback()
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/audio_helpers.py", line 288, in stop_playback
    self._source.stop()
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/audio_helpers.py", line 221, in stop
    self.flush()
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/audio_helpers.py", line 211, in flush
    self._audio_stream.write(b'\x00' * self._flush_size)
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 1202, in write
    _check(err)
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 2563, in _check
    raise PortAudioError(errormsg, err)
sounddevice.PortAudioError: Can't write to an input only stream [PaErrorCode -9974]

SoundGuy, 2018-02-05:
ok. it is now working. Is there any way to enable horword on PI b+? 
If not, anything I should know before trying either https://github.com/xtools-at/AssistantPi ? Does this work in tandem with that?

BadFlo, 2018-02-05:
If I'm not wrong, you can start Google-assistant service to get hotword, but as mentioned in the Readme, some functionalities will not work.
Also also, the log should be in /tmp folder.

SoundGuy, 2018-02-05:
starting the ok-google service seems to be doing nothing.

SoundGuy, 2018-02-05:
this is what i see in syslog:

Feb  5 07:01:22 piassistant systemd[1]: Started Google Assistant-Voice.
Feb  5 07:01:28 piassistant python[8182]: Traceback (most recent call last):
Feb  5 07:01:28 piassistant python[8182]:   File "/home/pi/GassistPi/src/main.py", line 29, in <module>
Feb  5 07:01:28 piassistant python[8182]:     from google.assistant.library import Assistant
Feb  5 07:01:28 piassistant python[8182]: ImportError: No module named 'google.assistant.library'
Feb  5 07:01:29 piassistant systemd[1]: gassistpi-ok-google.service: Main process exited, code=exited, status=1/FAILURE
Feb  5 07:01:29 piassistant systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Feb  5 07:01:29 piassistant systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
Feb  5 07:01:29 piassistant systemd[1]: gassistpi-ok-google.service: Service hold-off time over, scheduling restart.
Feb  5 07:01:29 piassistant systemd[1]: Stopped Google Assistant-Voice.
Feb  5 07:01:29 piassistant systemd[1]: Started Google Assistant-Voice.

shivasiddharth, 2018-02-05:
Pi B+ Does not support ok-google wakeword. It's not Armv7. You can only trigger it with a pushbutton. 
Regarding the portaudio error, it's one of the issues with the current API it has to be rectified by Google and not me. 
Regarding the xtools, you need to ask that to author of that project, they are the ones who can offer an answer.

closed 2018-02-05 by shivasiddharth

sunitasonu, 2019-10-21:
Exception:
Traceback (most recent call last):
  File "/home/pi/env/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connection.py", line 138, in _new_conn
    (self.host, self.port), self.timeout, **extra_kw)
  File "/home/pi/env/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/util/connection.py", line 98, in create_connection
    raise err
  File "/home/pi/env/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/util/connection.py", line 88, in create_connection
    sock.connect(sa)
OSError: [Errno 101] Network is unreachable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 594, in urlopen
    chunked=chunked)
  File "/home/pi/env/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 350, in _make_request
    self._validate_conn(conn)
  File "/home/pi/env/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 837, in _validate_conn
    conn.connect()
  File "/home/pi/env/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connection.py", line 281, in connect
    conn = self._new_conn()
  File "/home/pi/env/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connection.py", line 147, in _new_conn
    self, "Failed to establish a new connection: %s" % e)
requests.packages.urllib3.exceptions.NewConnectionError: <requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x76967130>: Failed to establish a new connection: [Errno 101] Network is unreachable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/home/pi/env/lib/python3.5/site-packages/pip/commands/install.py", line 353, in run
    wb.build(autobuilding=True)
  File "/home/pi/env/lib/python3.5/site-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/home/pi/env/lib/python3.5/site-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/home/pi/env/lib/python3.5/site-packages/pip/req/req_set.py", line 487, in _prepare_file
    req_to_install, finder)
  File "/home/pi/env/lib/python3.5/site-packages/pip/req/req_set.py", line 428, in _check_skip_installed
    req_to_install, upgrade_allowed)
  File "/home/pi/env/lib/python3.5/site-packages/pip/index.py", line 465, in find_requirement
    all_candidates = self.find_all_candidates(req.name)
  File "/home/pi/env/lib/python3.5/site-packages/pip/index.py", line 423, in find_all_candidates
    for page in self._get_pages(url_locations, project_name):
  File "/home/pi/env/lib/python3.5/site-packages/pip/index.py", line 568, in _get_pages
    page = self._get_page(location)
  File "/home/pi/env/lib/python3.5/site-packages/pip/index.py", line 683, in _get_page
    return HTMLPage.get_page(link, session=self.session)
  File "/home/pi/env/lib/python3.5/site-packages/pip/index.py", line 792, in get_page
    "Cache-Control": "max-age=600",
  File "/home/pi/env/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 501, in get
    return self.request('GET', url, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/pip/download.py", line 386, in request
    return super(PipSession, self).request(method, url, *args, **kwargs)
  File "/home/pi/env/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 488, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/pi/env/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 609, in send
    r = adapter.send(request, **kwargs)
  File "/home/pi/env/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/adapter.py", line 47, in send
    resp = super(CacheControlAdapter, self).send(request, **kw)
  File "/home/pi/env/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/adapters.py", line 423, in send
    timeout=timeout
  File "/home/pi/env/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 643, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "/home/pi/env/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/util/retry.py", line 315, in increment
    total -= 1
TypeError: unsupported operand type(s) for -=: 'Retry' and 'int'
I am getting this error plz any one can help me thank you

I