shivasiddharth/GassistPi #186: YouTube stream command crashes service

pmona, 2018-02-06:
Thank you Sid for the great project and help provided.

I have looked at the previously closed issue #163, but the fix there is already in place.
pi@raspberrypi:~ $ source env/bin/activate
(env) pi@raspberrypi:~ $ pip install youtube-dl
Requirement already satisfied: youtube-dl in ./env/lib/python3.5/site-packages
(env) pi@raspberrypi:~ $

The above is already in place yet, I continue to have a problem with the service crashing.  Any idea what is going on here?

pi@raspberrypi:~ $ sudo /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id 'pihome-186518'  --device_model_id 'Pi3'
device_model_id: Pi3
device_id: 98FC7E1B1C47C2DE851F06DA1E2201AC

https://embeddedassistant.googleapis.com/v1alpha2/projects/pihome-186518/devices/98FC7E1B1C47C2DE851F06DA1E2201AC 200
ON_MUTED_CHANGED:
  {'is_muted': False}
ON_START_FINISHED

ON_CONVERSATION_TURN_STARTED
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': 'stream Another Brick in the Wall'}
Getting youtube link
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of say.mp3 finished.
pi@raspberrypi:~ $

shivasiddharth, 2018-02-06:
Have you added the Youtube Data API Key ?

pmona, 2018-02-06:
Thank you, that was it.   I thought the API key was only needed for autoplay.

closed 2018-02-06 by pmona