shivasiddharth/GassistPi #187: Radio & Youtube: Audio issues

Mierola, 2018-02-06:
Hi!

I´m having audio issues when using the radio stream and youtube feature. Every minute or so I get distorted sound fow a few seconds.

Playing the same youtube videos in kodi works perfect.
Streaming radio through VLC (not through the assistant) also works perfectly.

Could I have missed a setting somewhere which only affect streaming through the assistant?

shivasiddharth, 2018-02-07:
There is really nothing much that can be done reg this issue. I am giving it a shot in the dark. You will have a file called mpv.conf in the /home/pi/.config/mpv directory open that and add a line
`cache=2048`.  Close save, restart and try playing something. Let me know as to how it goes either way.

Mierola, 2018-02-07:
Sorry, that didn´t work. I tried to increase the cache to be much bigger but the issue is still there.
Pointing out MPV as a potential cause of the issue was however interesting, I might search around and see if I can find anything that cause/solve this.

adas4190, 2018-02-08:
Same thing started for me after I added aiy voice hat. I fyou find any solution let me know pls.

shivasiddharth, 2018-02-08:
So is that mpv+voice hat issue ? @Mierola , May i know as to what is your Audio setup ?

Mierola, 2018-02-08:
I have a Rpi3+Voice Hat.
Would it be possible to change to vlc, or another audio/video player?

adas4190, 2018-02-08:
I have rpi3 + aiy voice hat. I used to have usb mic and jack output. The problem started after adding voice hat. I did clean install.

adas4190, 2018-02-08:
Music doesnt  seems to go slow or stop/ stutter it just go distorted for a while and then back to normal.

Mierola, 2018-02-08:
I updated the title since we now know the combination which result in this issues so others with the same issue can find it easier.

shivasiddharth, 2018-02-08:
@Mierola , By using VLC, we would loose volume control. The volume will not go down when assistant is awakened and we cannot change the media volume by voice. Also, VLC has some issues playing google music streams automatically.
First, we need to determine whether the distortion is at fixed intervals or is it a particular frequency in the audio that is causing the distortion. If you can run a stream using mpv from the terminal outside of the  assistant and if you can log the time as to when the distrotion is happening, we can dig further. 
I shall check if we can install some codecs which can help solve this issue.

Mierola, 2018-02-08:
OK, I have now played a short 5 min Youtube video twice, and a radio stream for at least 5 min in both lowest (32 kbps) and highest quality (192kbps) available.

First distortion was around 1:15-1:30.
After that each occurrence of the distortion is ~+1:00 (+/- 5s) after the previouc one.
Between these timings the sound was good and had no issues.
The distortion came both during regular talk and music, and no connection between the frequency of the noice/sound and the distortion was found.
I had no audio related errors as far as I could see, and the cache was at 10s+3MB during the entire stream. No cache drop during the distortion was spotted.

shivasiddharth, 2018-02-08:
@Mierola , Ok thanks. Let me experiment with some codecs and get back.

shivasiddharth, 2018-02-09:
Plugins dont seem to work.  Proceeding on to hardware testing.

Pnether, 2018-02-09:
I use a Seeed Respeaker 2-mic and had the same problem, but added a buffer to the mpv player:
Adding --audio-buffer .8 to the mpv player to increase the default buffer from 200ms to 800ms.
It resolved the stuttering not completely, but it has IMPROVED VERY MUCH after this change.

	cd /home/pi/GassistPi/src
	nano actions.py

	press <ctrl-w> to search for red header "-Start of Functions for YouTube Streaming-"

Below this header I added two times the --audio-buffer to the mpv player command:
     "os.system('mpv --really-quiet --volume='+str(startingvol)+' '+streamurl+' &')
into:
     "os.system('mpv --audio-buffer .8 --really-quiet --volume='+str(startingvol)+' '+streamurl+' &')

Maybe all occurences of "os.system('mpv" should be added with a buffer ? 
Did not try/test that, because I don't use all available functionalities.

With thanks to tyzoid at https://github.com/termux/termux-packages/issues/1127

cnoorman, 2018-02-10:
Aahh, luckily I'm not the only one with this problem.
I also have RPI3 & AIY voice hat. Identical problems. 
Distortion for 5 seconds every minute of audio playback with mpv. 
No problem when using vlc.
If you just play local mp3 files ( with all Gassist en Alexa services stopped) with: 
mpv --no-video *.mp3
the problem already occurs. So my conclusion is that it is not related to Sid's software. 
The fact that vlc on the voice hat works OK, excludes imo a hardware problem of the hat.
I don't know how the audio is routed to the hat. It goes at least through the alsa mixer. Apparently that combination is causing the distortion. 
Note that the mpv build is old (0.23.0) . I could not find a recent build for Raspberry.
BTW increasing audio buffer size doesn't help at all. Behavior remains identical. 

Who is going to solve this problem? 
It screws up SID's nice project completely.

Mierola, 2018-02-10:
I would rather sacrifice the voice-controlled media volume and use VLC than use the MPV-player with distortion on this setup.

How much work would it be to have both options in the code (and MPV as default) and let us comment out the option we don´t want to use?
Radio streaming, Youtube and the voice control looks fairly easy, but I notice some mpv code in the Google Music part. Do you use mpv for that too?

shivasiddharth, 2018-02-10:
@Mierola , I use mpv for all media streaming. I could not get vlc to play google music streams. Let me know if you can get vlc to play gmusic streams. 
Start the assistant manually, give a gmusic streaming command. then you will get a link on the terminal. Stop the playback using ctrl+c and see if vlc plays the link that you just got.

Mierola, 2018-02-10:
No, it doesn´t work for me either.
And it´s probably no surprise that I get the the distortion when I´m playing songs from Google music too.

shivasiddharth, 2018-02-10:
Can any of you try playing these two streams using mpv in terminal and let me know if you get that distortion in either of the streams:
```
http://streaming.shoutcast.com/80sPlanet?lang=en-US
http://swr-swr1-bw.cast.addradio.de/swr/swr1/bw/mp3/64/stream.mp3
```

Mierola, 2018-02-10:
Distortion on both.

cnoorman, 2018-02-10:
As I mentioned above you don't have to stream. Just playing an mp3 file is enough to get the distortion.
What is the difference between the google AIY hat and the custom hat you are using Sid? Because aparantly that one is not showing the distortion.

Mierola, 2018-02-10:
Most of the broken features can be replaced by using existing Kodi commands instead, except for the radio streaming.
Kodi have some radio plugins, would it be possible to add command for using them as well?

shivasiddharth, 2018-02-10:
@cnoorman , Not much, Its just that, i have not used the MOSFET drivers for servo controls, pretty much everything else is the same, same audio amp and same mics as well and i do get the same distortion on custom HAT, its not noticeable in the videos.
@Mierola , relax a bit, 3 secs of distortion don't make the features broken. I am checking to see if the MAX98357A has any role to play. The clock pulses are known to go out of phase in MAX98357A audio amp. I am building a second custom HAT with interchangeable DACs to see if its the Google's drivers or if its the DAC itself.

cnoorman, 2018-02-10:
OK, I didn't realize that you had the same problems with the custom HAT. The MAX being the culprit seems unlikely to me since it works fine with VLC.

Pnether, 2018-02-10:
Found a way to play both streams on my Pi model B with Seeed Respeaker 2-mic
with **VERY MUCH** better playback, just by trial and error and playing with the mpv options on
https://github.com/mpv-player/mpv/blob/master/DOCS/man/options.rst

`mpv --alsa-resample=yes --audio-buffer .8 http://streaming.shoutcast.com/80sPlanet?lang=en-US`
`mpv --alsa-resample=yes --audio-buffer .8 http://swr-swr1-bw.cast.addradio.de/swr/swr1/bw/mp3/64/stream.mp3`

The `--alsa-resample=yes` was really necessary, I am not sure of improvement by `--audio-buffer .8`

shivasiddharth, 2018-02-11:
Here is the fix:
The mp3 codecs installed along with mpg123 and vlc were causing the issue. So to fix the stuttering, we need to uninstall/remove both.
```
sudo apt-get remove mpg123 -y
sudo apt-get remove vlc -y
sudo apt-get purge
sudo apt autoremove
```
In the actions.py file scroll to the function say and in there, change line` os.system("mpg123 "+ttsfilename)` to `os.system("mplayer "+ttsfilename)`

This fixed the issue for me for both the AIY and custom HAT, let me know if it worked for you ppl as well.

Pnether, 2018-02-11:
Pi3B with Seeed Respeaker 2-mic: followed remove / edit actions.py, rebooted, and few minutes listening:
- Link1  US: still many sound dropouts, 
          NOTE: with `mpv --alsa-resample=yes` there are **NO dropouts** at all for me.
- Link2 mp3: **NO dropouts** at all, but maybe some minimal ticks/distortion in the sound ?
          AGAIN: with `mpv --alsa-resample=yes` it SEEMS more perfect for me.

cnoorman, 2018-02-11:
Unfortunately removal of mpg123 and vlc does not solve the problem on Pi3+Google HAT. 
Distortion is the same, however the occurrence rate has changed. 
First occurrence now after ~2 min (was ~1min), after that every ~1.5 min (before ~1 min) . Length still about 5 secs. So the characteristics changed but the effect is still very much there.
I note that the phenomenon is independent of the activity within mpv. It doesn't matter if you play a long stream or separate tracks. There is no relation with the starting moment of a new track, only with the previous moment of distortion.
NB audio-buffer size and alsa-resample on/off have no effect in case of Google HAT.
I would focus now first on the Google HAT driver.....

Enzoinventor, 2018-02-11:
I also distort myself with the aiy and dac google kits

Enzoinventor, 2018-02-11:
this seems to be going very well:
sudo apt-get install mpg321
In the actions.py file, scroll down to the function to say and in there, change line os.system ("mpg123" + ttsfilename) aos.system ("mpg321" + ttsfilename)
let me know!!!

Pnether, 2018-02-11:
@Enzoinventor
The line with os.system ("xxxxxx" + ttsfilename) only involves the say function (text to speech), not the url streaming function by mpv. Nevertheless, for me the mpg321 spoken text is better than mplayer, which spoke the text in a partial broken way. But not better than the original mpg123 player.

shivasiddharth, 2018-02-11:
Just curious, all those having this issues, did you ppl get the HAT in last May (Magpi 57) or thereafter during the public release?

cnoorman, 2018-02-11:
I got my Google AIY Projects Voice Kit (including HAT) from the public release (a month ago).
Started this afternoon with fresh Raspbian Linux 9 (stretch) install (aiyprojects-2018-01-03.img.xz)  
After that I did:

sudo apt-get update
sudo apt-get install libxml2-dev
sudo apt-get install libxslt-dev
sudo apt-get install mpv
sudo cp ./AIY-projects-python/scripts/asound.conf ./.asoundrc

So  the minimal setup for the mpv player. In this state the distortion is present when just playing mp3: files: 
mpv --no-video *.mp3

Maybe we should raise an issue at Raspberry or MagPi??

BadFlo, 2018-02-11:
Can you try with standard Raspbian image (non-aiy) as specified in Readme?

shivasiddharth, 2018-02-11:
@BadFlo, We are just trying to resolve an issue here with AIY HAT... Not connected to the project.

BadFlo, 2018-02-11:
Yes, but is the issue the same with standard Raspbian image? Maybe it's
related to the aiy project image.

TedMichalik, 2018-02-11:
I have the AIY HAT from the public release and use the standard Raspbian desktop image. Still getting distorted audio after purging mpg123 and vlc, and installing mpg321.

shivasiddharth, 2018-02-11:
@BadFlo The issue is with All of the Raspbian images.

shivasiddharth, 2018-02-11:
@cnoorman , As it is we cannot open "issues" anywhere. AIY folks would dismiss this as mpv issue and mpv folks would dismiss this as AIY HAT issue. The mpv package on debian is 0.23.0, while mpv folks have made a good number of improvements since then. So ideal solution would be to seek help to install the latest mpv on raspbian. This post has to be made in Raspberry Pi forums.
This however seems more like an AIY HAT issue. 
Here is the link to the AIY HAT codec. https://github.com/raspberrypi/linux/blob/rpi-4.9.y/sound/soc/bcm/googlevoicehat-codec.c 
The `dmesg | grep voicehat` reads the following:
```
[    3.151077] snd-googlevoicehat-soundcard soc:sound: ASoC: CODEC DAI voicehat-hifi not registered - will retry
[    3.151096] snd-googlevoicehat-soundcard soc:sound: snd_soc_register_card() failed: -517
[    3.153287] voicehat-codec voicehat-codec: property 'voicehat_sdmode_delay' not found default 5 mS
[    3.163350] snd-googlevoicehat-soundcard soc:sound: voicehat-hifi <-> 3f203000.i2s mapping ok
```
For some reason, its failing to register the card/codec.

shivasiddharth, 2018-02-11:
The issue could be related to the `softvol` settings for the AIY HAT. If you change your asound.conf and .asoundrc to the following, then the crackling doe not appear, but at the same time, you will loose the volume control, the speaker volume will be set to maximum. 
```
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:0,0"
        channels 2
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:0,0"
        }

```
This seems more like an AIY HAT issue than anything else.

cnoorman, 2018-02-12:
@shivasiddharth 
For me `dmesg | grep` voicehat reads the following:
```
[    3.231554] voicehat-codec voicehat-codec: property 'voicehat_sdmode_delay' not found default 5 mS
[    3.449284] snd-googlevoicehat-soundcard soc:sound: voicehat-hifi <-> 3f203000.i2s mapping ok
[   25.545628] voicehat-codec voicehat-codec: Enabling audio amp...
[   31.284462] voicehat-codec voicehat-codec: Disabling audio amp...
[   31.730867] voicehat-codec voicehat-codec: Enabling audio amp...
[   31.732417] voicehat-codec voicehat-codec: Disabling audio amp...
.....
```
The `snd-googlevoicehat-soundcard soc:sound` lines don't show up here.
Enabling/disabling of codec goes on continuously while the music is playing. Is that normal?

I confirm that the original problem is solved when changeing asound.conf and .asoundrc files as given above (no softvol). Note that at the bottom one '}' is missing. 
So maybe one step closer to really solving the mistery.
Adding the `mpv --volume=25` does not prevent the prompting and GA voice to be blaring, so it's not really a solution.

shivasiddharth, 2018-02-12:
@cnoorman , Pulseaudio will solve this issue, but the audio quality will degrade a bit. The other option is to switch to VLC which does not make sense given that the gmusic streams do not work in VLC. And 🤣  omxplayer does nothing other than wishing a nice day.

cnoorman, 2018-02-12:
@shivasiddharth , Will the GM stream work in Pulseaudio? Little bit of degradation would be OK on a one speaker cardboard box :-).
In the mean time I build mpv-0.27.0 . That also doesn'help. Same distortion behaviour.

shivasiddharth, 2018-02-12:
@cnoorman , This issue is pretty much common in all of the Pi Voice HATs (Respeaker 2Mic, AIY) that share the I2S for Mic and speaker. Almost all of the budget SBCs have single I2S channel, new SBCs are coming up with multiple I2S channels but prices will be upwards of $100.

I just tried the pulseaudio, it has a bit of static but not the every minute garbling. So those who are interested in trying:
1. Restore the .asoundrc and asound.conf of AIY just in case if you have changed anything.
2. Install pulseaudio and pulse audio control using:
```
sudo apt-get update
sudo apt-get install pulseaudio
sudo apt-get install pavucontrol

```
Pulse audio control is required as the volume control on the menubar and alsa will be disabled upon pulse audio installation.
3. Start the assistant either manually or the service and then open the pulse audio control by executing the following in a terminal:
`pavucontrol`
4. There adjust the volume under playback.

There is nothing much more that i can do in regards to this issue. There is infact one last thing that you can do if you wish to.
Detach the Mic from the AIY HAT. Connect it directly to the Pi, compile I2S kernels and use that for audio input and the Pi's jack for audio out. If you wish to do that then follow this guide from Adafruit (https://learn.adafruit.com/adafruit-i2s-mems-microphone-breakout/raspberry-pi-wiring-and-test)

cnoorman, 2018-02-14:
@shivasiddharth , I played around shortly with pulseaudio / pavucontrol. Works well with GA, but not with Alexa. I didn't put further time in it. 

As you mentioned before the problem has to do with the `softvol` feauture from ALSA. Instead of taking out `softvol` completely you can also add the following --audio-device  parameter to mpv:

mpv `--audio-device=alsa/plughw:CARD=sndrpigooglevoi,DEV=0` --volume=25 --no-video *.mp3

This way you bypass the `softvol` control in ALSA only for mpv. The rest of the volume control remains available. Still not perfect, but better than taking out `softvol` completely.

Sharing I2S for Mic and speaker may not be optimal, but works good enough for these kind of applications (as long as you don't need mpv!).
Detaching the Mic from the AIY HAT is not a good idea. I'd rather use a 2$ usb Mic instead. Also it takes the fun out of the original card-board box concept.

I will continue to try and find the reason for difference in performance between vlc and mpv, Initially focussing on ALSA. At some point I may open an issue at an AIY or Raspberry forum.
Will keep you posted on any progress.

shivasiddharth, 2018-02-15:
@cnoorman , I had looked at this before. The problem is, with `alsa/plughw:`, the dmixer will not work and without the dmixer, when music is playing, assistant will not work. But if you change `alsa/plughw:` to `alsa/dmix:`, the assistant can also be used, but the distortion becomes a continuous occurrence. The ideal solution at this point would be to use the pulseaudio.

shivasiddharth, 2018-02-23:
@Mierola , Do we have an update on the status of this issue ? Whether you tried the pulseaudio or if you have moved on?

Mierola, 2018-02-23:
I have actually bought a Google Home Mini, so not much time spent on this issue the last week.

aandroide, 2018-02-24:
a new vlc update has been released a few days ago: https: //www.videolan.org/vlc/releases/3.0.0.html
we just have to hope that we can import it on the raspberry
with vlc aiy sound users had no problems

aandroide, 2018-02-24:
is it possible for users of AIY-HAT users, to use the card's microphone and to use the audio output jack?
Could mpv stuttering be avoided?

patreiche, 2018-03-18:
I also have the same problem. I just installed mine and have it operational, thank you for all your hard work. 
I remember reading about codecs for sale on here when researching the AYI HAT project. I think it is a codec issue but I was glad to see that it is not only me that has this issue :).

shivasiddharth, 2018-06-02:
This issue has been fixed py porting the project to python-vlc

PR #371 (2018-06-02): Moham gassistpi
Fixes #187 
Ported the media player to python-vlc. 
Simplified YouTube and Google music functions.
Unified Music player with better control, supports automatic track switching after the end of first track.
closed 2018-06-02 by shivasiddharth