shivasiddharth/GassistPi #188: sounddevice.PortAudioError: Error opening RawStream: Device unavailable [PaErrorCode -9985]

moham96, 2018-02-06:
Hi,
On the raspberry pi 3 i have this issue where the main.py works while pushbutton.py crashes with the following error

> 2018-02-05 12:18:40,192 DEBUG google.auth.transport.requests Making request: POST https://accounts.google.com/o/oauth2/token
> 2018-02-05 12:18:40,208 DEBUG requests.packages.urllib3.connectionpool Starting new HTTPS connection (1): accounts.google.com
> 2018-02-05 12:18:40,740 DEBUG requests.packages.urllib3.connectionpool https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
> 2018-02-05 12:18:40,781 INFO root Connecting to embeddedassistant.googleapis.com
> 2018-02-05 12:18:41,894 ERROR __main__ Error opening RawStream: Device unavailable [PaErrorCode -9985]
> Traceback (most recent call last):
>   File "./GassistPi/src/pushbutton.py", line 818, in <module>
>     main()
>   File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 722, in __call__
>     return self.main(*args, **kwargs)
>   File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 697, in main
>     rv = self.invoke(ctx)
>   File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 895, in invoke
>     return ctx.invoke(self.callback, **ctx.params)
>   File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 535, in invoke
>     return callback(*args, **kwargs)
>   File "./GassistPi/src/pushbutton.py", line 709, in main
>     flush_size=audio_flush_size
>   File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/audio_helpers.py", line 187, in __init__
>     blocksize=int(block_size/2),  # blocksize is in number of frames.
>   File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 1256, in __init__
>     **_remove_self(locals()))
>   File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 772, in __init__
>     'Error opening {0}'.format(self.__class__.__name__))
>   File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 2563, in _check
>     raise PortAudioError(errormsg, err)
> sounddevice.PortAudioError: Error opening RawStream: Device unavailable [PaErrorCode -9985]

shivasiddharth, 2018-02-06:
Check this [out](https://github.com/shivasiddharth/GassistPi/issues/182#issuecomment-363006997) Its a google issue. The timing for audio input has to be precise else it will throw-out this error. If you give the voice command a bit early or a bit late, this happens. This error occurs even with google's unmodified pushtotalk. This is exactly the reason why i removed the custom wakeword.

closed 2018-02-06 by shivasiddharth

moham96, 2018-02-06:
@shivasiddharth  Did you report the issue to google team?

shivasiddharth, 2018-02-06:
Someone has already done it https://github.com/googlesamples/assistant-sdk-python/issues/155