shivasiddharth/GassistPi #191: SoundDeviceStream read overflow

mechaturtles, 2018-02-07:
My current setup is a Raspberry Pi 1 B+ with the AIY Voice Kit. 
I actually received the error before, so I thought that reinstalling GassistPi would work. I was able to get the light, mic, and sounds to work, and I would receive the startup sound. However, the program was always on/listening, the button didn't even work, and the requests would suddenly cut off (e.g., "What's the time?" Response: "The time is six forty-*program crashes*").

The speaker is confirmed to be working on this install, but I can't get anything else to work.
I received the following error:

INFO:root:Connecting to embeddedassistant.googleapis.com
Press Enter to send a new request...
INFO:root:Recording audio request.
WARNING:root:SoundDeviceStream read overflow (3200, 6400)
INFO:root:End of audio request detected
INFO:root:Finished playing assistant response.
Press Enter to send a new request...
INFO:root:Recording audio request.
WARNING:root:SoundDeviceStream read overflow (3200, 6400)
INFO:root:End of audio request detected
INFO:root:Finished playing assistant response.
Traceback (most recent call last):
  File "/home/pi/env/bin/googlesamples-assistant-pushtotalk", line 11, in <module>
    sys.exit(main())
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/pushtotalk.py", line 424, in main
    continue_conversation = assistant.assist()
  File "/home/pi/env/lib/python3.5/site-packages/tenacity/__init__.py", line 214, in wrapped_f
    return self.call(f, *args, **kw)
  File "/home/pi/env/lib/python3.5/site-packages/tenacity/__init__.py", line 295, in call
    start_time=start_time)
  File "/home/pi/env/lib/python3.5/site-packages/tenacity/__init__.py", line 252, in iter
    return fut.result()
  File "/usr/lib/python3.5/concurrent/futures/_base.py", line 398, in result
    return self.__get_result()
  File "/usr/lib/python3.5/concurrent/futures/_base.py", line 357, in __get_result
    raise self._exception
  File "/home/pi/env/lib/python3.5/site-packages/tenacity/__init__.py", line 298, in call
    result = fn(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/pushtotalk.py", line 169, in assist
    self.conversation_stream.stop_playback()
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/audio_helpers.py", line 288, in stop_playback
    self._source.stop()
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/audio_helpers.py", line 221, in stop
    self.flush()
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/audio_helpers.py", line 211, in flush
    self._audio_stream.write(b'\x00' * self._flush_size)
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 1202, in write
    _check(err)
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 2563, in _check
    raise PortAudioError(errormsg, err)
sounddevice.PortAudioError: Can't write to an input only stream [PaErrorCode -9974]

shivasiddharth, 2018-02-07:
All the errors associated with the pushbutton.py will be fixed in the next update. However, this Portaudio error needs to be fixed from google's side. This occurs if the voice commands are either slightly late or slightly early, i dont have control over that error.

MartinMajewski, 2018-03-01:
And now? I'm getting the same error while installing on a Raspberry Pi Zero W.

`INFO:root:Registering device
INFO:root:Device registered: 1ae438a2-1d79-11e8-9f0c-b827eba3f494
Press Enter to send a new request...
INFO:root:Recording audio request.
WARNING:root:SoundDeviceStream read overflow (3200, 6400)
WARNING:root:SoundDeviceStream read overflow (3200, 6400)
INFO:root:End of audio request detected
INFO:root:Finished playing assistant response.
Traceback (most recent call last):
  File "/home/pi/env/bin/googlesamples-assistant-pushtotalk", line 11, in <module>
    sys.exit(main())
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/pushtotalk.py", line 424, in main
    continue_conversation = assistant.assist()
  File "/home/pi/env/lib/python3.5/site-packages/tenacity/__init__.py", line 214, in wrapped_f
    return self.call(f, *args, **kw)
  File "/home/pi/env/lib/python3.5/site-packages/tenacity/__init__.py", line 295, in call
    start_time=start_time)
  File "/home/pi/env/lib/python3.5/site-packages/tenacity/__init__.py", line 252, in iter
    return fut.result()
  File "/usr/lib/python3.5/concurrent/futures/_base.py", line 398, in result
    return self.__get_result()
  File "/usr/lib/python3.5/concurrent/futures/_base.py", line 357, in __get_result
    raise self._exception
  File "/home/pi/env/lib/python3.5/site-packages/tenacity/__init__.py", line 298, in call
    result = fn(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/pushtotalk.py", line 169, in assist
    self.conversation_stream.stop_playback()
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/audio_helpers.py", line 288, in stop_playback
    self._source.stop()
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/audio_helpers.py", line 221, in stop
    self.flush()
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/audio_helpers.py", line 211, in flush
    self._audio_stream.write(b'\x00' * self._flush_size)
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 1202, in write
    _check(err)
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 2563, in _check
    raise PortAudioError(errormsg, err)
sounddevice.PortAudioError: Can't write to an input only stream [PaErrorCode -9974]`

shivasiddharth, 2018-03-18:
Port audio error is a google's issue please refer to https://github.com/googlesamples/assistant-sdk-python/issues/155

closed 2018-03-18 by shivasiddharth