shivasiddharth/GassistPi #193: New update guide pls

adas4190, 2018-02-09:
Could You please make up to date guide how to use your upgrade script. Im so sorry but I am total noob and I have messed up my instalation somehow when I tried to update it as you show in previous videos. 
Thank You.

shivasiddharth, 2018-02-15:
You need to make the update script executable using chmod command and then execute, its easy and straight forward.

jabulonboaz, 2018-02-15:
@shivasiddharth i do not know where i can ask this question so i place it here and please feel free to point me to the right direction.

How can i point GA to run a script i have instead of say triggering a GPIO pin in the actions.py file please?

adas4190, 2018-02-15:
When I did as you said (I have make script executable) after I run it assistant didn't work. Maybe I have messed up something. Should I comment enything? I'm using raspberry pi 3

shivasiddharth, 2018-02-16:
@adas4190 , after updating, you still need to add back those credentials in actions.py. pls attach the log, only then i can see the issue, you can find the log in /tmp directory.

shivasiddharth, 2018-03-03:
Closing this for inactivity

closed 2018-03-03 by shivasiddharth

celispin, 2018-10-01:
Hello, is there any updated guide on how to updated GassistPi?

V1JCM, 2019-01-19:
Update script not available, in GassistPi/scripts there are only 
gassist-installer.sh
service-installer.sh
There is no update.sh in that directory. Where can I get one?