shivasiddharth/GassistPi #194: How to print the recognised text by Google Assistant in a terminal window ?

Pnether, 2018-02-09:
During installation, everything that after Hey Google is printed out in a terminal window.
Can this text can also be made visible AFTER headless installation of the assistant ?

Reason is, that I have big trouble with selecting proper keywords that always will be recognised correctly.
(eg. the keyword "trigger" seems not to function for me, 
 and "shutdown" without a space works better than "shut down")

Does anybody know how to make the spoken words visual in printed text ?

BadFlo, 2018-02-09:
You can stop the service and start it manually on a cmd with the appropriate line in the service file under systemd folder.

Pnether, 2018-02-09:
Since I am not a very experienced Linux user, your answer needed some trial and error from my side,
but it worked fine ! Thanks, Badflo !

For other inexperienced users I will write your solution down in more detail:
        `sudo systemctl stop gassistpi-ok-google.service`     (this is the service that I use)
	`cd /home/pi/GassistPi/systemd`
	`nano gassistpi-ok-google.service.sh`

Find the following line (note: exchange xxxxx and yyyyy with your personal credentials):
	`ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id xxxxx  --device_model_id marine-copilot-xxxxx-yyyyy`

In a terminal, copy from line above and paste:
	`/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id xxxxx  --device_model_id marine-copilot-xxxxx-yyyyy`

shivasiddharth, 2018-02-09:
I am considering this issue resolved and closing this !

closed 2018-02-09 by shivasiddharth

shivasiddharth, 2018-02-10:
@Pnether ,
I felt that there was a need to mention this just for the benefit of new users, that is why i have added a section in the readme doc (https://github.com/shivasiddharth/GassistPi/blob/master/README.md#manually-start-the-assistant) and hence closed this issue.

Pnether, 2018-02-10:
> master/README.md#manually-start-the-assistant) and hence closed this

That is great, Shivasiddharth !

Thanks ! and have a nice weekend !

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virusvrij.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>