shivasiddharth/GassistPi #195: Bluetooth speaker and USB microphone?

defar123, 2018-02-10:
Hi
Is it possible to use Bluetooth speaker and usb microphone with this? I can't figure out how to

shivasiddharth, 2018-02-15:
Sure, you can there are some guides in raspberry pi forums for pairing bluetooth speaker, make sure to go through that.

chsims1, 2018-02-17:
I found this guide to be useful https://sigmdel.ca/michel/ha/rpi/bluetooth_01_en.html

legemulo, 2018-02-18:
The thing is after a long battle I successfully got it to work but Google Assistant stopped working, that is, I don't know how to configure the files for his voice come through the bluetooth speakers. Any help?

shivasiddharth, 2018-03-18:
Closing this for inactivity.

closed 2018-03-18 by shivasiddharth

Vino-git, 2018-08-21:
Hi, I have Bluetooth speaker working with rasp-zero... Now I want google home to speak through my paired Bluetooth speaker. Pls, suggest what should be done.