shivasiddharth/GassistPi #196: GPIO 22 Manual Triggering not working

The-MAI, 2018-02-10:
Hi !!

GPIO 22 connected to LOW or HIGH -- doesnot trigger the Google Assistant.

Is there any step that is being missed or any change in the configuration files??

Enzoinventor, 2018-02-10:
#190 
He answered me this way but I did not solve the problem

shivasiddharth, 2018-02-10:
@The-MAI Can you please attach the log file here and post the status of `sudo systemctl status gassistpi-push-button.service`
@Enzoinventor All that i can do is implement the fix and test if it works. If it works then i make it public. There could be so many user related issues like connecting a pushbutton between 22 and 3.3V instead of 22 and Grnd or connecting a pushbutton on Pin 22 instead of GPIO 22.

Enzoinventor, 2018-02-10:
pi@raspberrypi:~ $ sudo systemctl status gassistpi-push-button.service
● gassistpi-push-button.service - Google Assistant-Pushbutton
   Loaded: loaded (/lib/systemd/system/gassistpi-push-button.service; enabled; v
   Active: active (running) since Sat 2018-02-10 15:13:51 CET; 1min 15s ago
 Main PID: 444 (python)
   CGroup: /system.slice/gassistpi-push-button.service
           └─444 /home/pi/env/bin/python -u /home/pi/GassistPi/src/pushbutton.py

feb 10 15:13:51 raspberrypi systemd[1]: Started Google Assistant-Pushbutton.
feb 10 15:14:51 raspberrypi python[444]: buongiorno
lines 1-9/9 (END)

closed 2018-03-02 by shivasiddharth

aandroide, 2018-03-18:
#271  read here

commit 6865d326 (2018-03-02): New Features and Fixes (#230)

Fixes #233 Fixes #79 Fixes #196 Fixes #228 Fixes #229 

Feature additions:
1. Create a emulated philips Hue bridge service.  
2. Turn on/off, change colour, brightness of emulated philips hue lights. 
3. Get recipe info pushed to mobile/tablet.  
4. Track kickstarter campaigns by voice.
5. Remote control of Magic Mirror.