shivasiddharth/GassistPi #199: Question: How can I enable safe shutdown via hotword ?

jabulonboaz, 2018-02-11:
How can I trigger voice activated shutdown please so I don’t have to turn the power off abruptly and damage the pi and the SD card ? I.e “Alexa or hey google shutdown”

shivasiddharth, 2018-02-11:
I invested so much of time in drafting this extensive README and yet people dont read that "https://github.com/shivasiddharth/GassistPi#voice-control-of-gpios-servo-and-pi-shutdown"

closed 2018-02-11 by shivasiddharth

jabulonboaz, 2018-02-11:
@shivasiddharth thank you apologies did not spot it before.

jabulonboaz, 2018-02-11:
@shivasiddharth i tried the instructions but its not working. You mentioned that by default the word is "trigger shutdown". When i say alexa or hey google trigger shutdown i get "sorry i don't see that iftt trigger" from alexa and "sorry i dont know how to help with that"

What am i doing wrong?

shivasiddharth, 2018-02-11:
These custom commands dont work with Alexa. If GA says  "sorry i dont know how to help with that" it means, it did not pickup the keyword "trigger". Manually start the assistant and see as to what it picks up your command as.

jabulonboaz, 2018-02-11:
I tried again and again it wont shut down. It will ether not respond or say what i wrote before but it wont shutdown the pi.  Could you please kindly  remind me how do i start GA manually so i can check?

shivasiddharth, 2018-02-11:
Again the info you are requesting is there in README: https://github.com/shivasiddharth/GassistPi/blob/master/README.md#manually-start-the-assistant

jabulonboaz, 2018-02-11:
These instructions are for Giassitpi i have not implemented that. I have assitantspi your other project. Are these the same?

shivasiddharth, 2018-02-11:
https://github.com/shivasiddharth/Assistants-Pi/blob/master/README.md#note

jabulonboaz, 2018-02-11:
@shivasiddharth thanks for posting this link but i have read it before. Sorry i can't seem find the part where you mention how to run the GA manually in AssistantsPi.

jabulonboaz, 2018-02-11:
@shivasiddharth or @channel or @all 
How can i manually trigger GA on AssistansPi for debugging the voice comands please?

Could it be that the features such as voice triggered shudown supported on GassistPi are broken or not supported  in AssistansPi ? 
Maybe i need to somehow update the build by pulling something with some command form the GassitPi git? 

This is a 2 week old AssistantsPi build on a pi3vb and it was done after @shivasiddharth recently fixed the issues that prevented AVS to authenticate. 
 
I also like to youse the GPIO triggers with LEDs or relays locally like in the GassistPi video of @shivasiddharth but i still like to use AssistantsPi.

jabulonboaz, 2018-02-11:
Anyone please?