shivasiddharth/GassistPi #200: How to trigger the Respeaker 2mic hat LEDs when Alexa and/or Google Assitant listen and respond ?

jabulonboaz, 2018-02-11:
The 3 LEDs of the Respeaker 2mic hat are controlled via SPI that I only now started to read about  here https://github.com/tinue/APA102_Pi
This is from the Respeaker product description:
http://wiki.seeed.cc/Respeaker_2_Mics_Pi_HAT/
‘RGB LED: 3 APA102 RGB LEDs, connected to SPI interface
I did try to change the GPIO settings in the AlexaPi relevant raspberry pi LED section at “config .yaml”  but I do not know how to do this on AssistansPi and I did successfully enabled the Respeaker button on GPIO 17 for AlexaPi  triggering by pressing it in addition to the hotword activation but when I change the LED settings by inserting GPIO 10 that is the respeaker pin for its rgb leds nothing happens. 

AlexaPi config.yaml alterations:
sudo nano /etc/opt/AlexaPi/config.yaml

voice_confirm: true

  raspberrypi:
    # GPIO Pin with button connected
    button: 17 # GPIO 17 corresponds to the respeaker 2mic hat push button 
    # GPIO Pin for the playback/activity light
    plb_light: 10 # Read below 
    # GPIO Pin for the recording light
    rec_light: 10 # GPIO 10 is the data pin of the Respeaker 2mic Hat  3 LEDs also tried 11 that’s the CLK of these LEDs

Is there a similar config for AssistantsPi please do activate the botton and the LEDs - if not the button as I recall there where some issues with this trigger in AssistansPi then just the LEDs?

I am just an armature with an engineering background trying to learn by googling and reading the great work that yourself and others are putting together I am not a coder or a developer I am afraid .
Any pointers, any URLs or any tips you can recommend I am happy to follow. 

Other questions:
Another thing I like to do is to be able to control with buttons connected to GPIO pins the volume of Alsa mixer again any tips and online material you can point too I would be very grateful. 
Also is it possible to enable a voice trigger shutdown ? I saw @shivasiddharth doing it in his videos on YouTube but never understood how to activate this feature.
Finally I am trying to understand how to setup custome wakewords or hotword using Snowboy so any pointers on how exactly to modify what part of the AssistantsPi code to do this I would appreciate a lot too. 
Thank you in advance

shivasiddharth, 2018-02-11:
This question should be in seed studio's git. But anyways, they have a section in their wiki: http://wiki.seeed.cc/Respeaker_2_Mics_Pi_HAT/#how-to-use-the-on-board-apa102-leds if you visit the git, they have even more details and even examples for google assistant.

And, this place is not mean for generalized queries. You have the Raspbian forums for a reason. if your search for "Controlling alsa using GPIO" ton of information will pop up.

There is already an ongoing conversation about custom wakeword and i have clearly given out the reasons for removing it.

closed 2018-02-11 by shivasiddharth

jabulonboaz, 2018-02-11:
How is this a generalised question when i am trying to understand how to do this in AssistantsPi? 
I am just trying to learn here.