shivasiddharth/GassistPi #201: YouTube api and Google Play Music app password

legemulo, 2018-02-11:
Hello! Thank you so much for this fantastic job! After so much research I finally got the Googe assistant to work on a headless raspberry pi! However, when I edited the actions.py with YouTube API and Google Play Music app password but I got an errorname not defined and the assistant would not start. Could you kindly help?

shivasiddharth, 2018-02-11:
From what i guess, you could have forgotten to enclose them within single quotes ('). If you cna attach the log file, i can for sure, say the reason. The log file can be found in /tmp folder.

legemulo, 2018-02-11:
That was it. I have just tried the feature and i is working! Thanks a lot! Sorry you have to deal with dummies like me!

shivasiddharth, 2018-02-11:
Closing this issue then..

closed 2018-02-11 by shivasiddharth