shivasiddharth/GassistPi #202: Device cannot be seen in app

dhruvwagh, 2018-02-11:
Hey I followed all the procedures and Google assistant works well on rpi3. O can also see the activities on the activity page but I'm unable to find the device as virtual device in Google home app nor in the assistant app. I'm from India I hope this is not because of the region please help..

Enzoinventor, 2018-02-11:
even I myself problem

albzn540, 2018-02-12:
This is because Google has limited the developer package that is available for installing on the Raspberry pi. There's no work around it. To see the device in the Google Home app you need to buy an actual google home.

shivasiddharth, 2018-02-12:
@dhruvwagh , You need to have a phone running minimum of Android N . I have shown here as to where you cna find the device https://youtu.be/vprJBDDYE8Q?t=997

dhruvwagh, 2018-02-13:
I have Android 8.1 that doesn't seem to be an issue. I changed my language from English India to English US an voila!! it worked. However I Have trouble changing the nickname and the room the device is in. Gives me some error saying cannot connect please try again later.

closed 2018-02-13 by dhruvwagh

shivasiddharth, 2018-02-13:
There are two places where your device will appear. It will not permit you to change the device name under Home Control. The language Selection and name change has to be under the devices option.

aandroide, 2018-02-21:
gentlemen, I discovered the problem
for those who live in Italy, you have to change the language in the phone settings and put English, only in this way by opening the app assistant and going to settings you see the registered devices.
However, the problem remains that the Italian language is not yet available.