shivasiddharth/GassistPi #203: IR support for Node MCU

shan2907, 2018-02-11:
it would be cool if there was a feature to send IR commands via the Node MCU. Ability to control not just switches but devices like T.V ,A.C. would be amazing!

Pnether, 2018-02-16:
Controlling my audio/TV equipment also was one of my goals. Instead I bought a Harmony Hub to be able to do this, because of it's huge library with IR codes. Switching-on with an activity command the TV, amplifier, settopbox, all with correct inputs in one time is no problem. But afterwards it became apparent that the Harmony Hub was only able to control Activities, not individual apparatus the way I wanted. 

Therefore I tried LIRC on a Pi Zero W and put a IR receiver / transmitter hat on it. As proof of concept, I only implemented the TV On/Off command in the Zero W, and used GassistPi on Pi3B to command the Zero W to send the IR code for TV On/Off. And this worked well. 

But the single LED transmitter is not very powerful when compared to the Harmony Hub's transmitters, and no IR library for most of my equipment (so a lot of work to read in all codes to begin with). 

I also wonder how many keywords would be involved and if the assistant is able to recognise them all correctly ? I tried quite a quite few keywords, but some resulted in many misunderstandings by the assistant and some just could not be used for this reason. Sometimes I really wonder if there is some recognition problem in my assistant or hardware like the microphone.

shivasiddharth, 2018-02-16:
@Pnether , Its really not cost effective to implement IR control of devices. As IR is line of sight, you need to have at-least one ESP/device. And most of the TVs change channels by codes, i dont think any one would have patience to map all their IR remote's keys.
I will provide a sample ESP Code and a sample of GassisPi ESP interaction, from there its upto the user to expand. I dont have a TV, so i cannot dig much into this.

Pnether, 2018-02-17:
@shivasiddharth: Very interesting to read you don't have a TV. Maybe too personal question, maybe not: is that for principal reasons, you don't want to waste time, or the GassistPi project eats all your time ?
Also, I want to mention link [https://github.com/sushilks/harmonyHubCLI](url)
I installed the CLI on a second Pi Zero W and have send commands to the Zero with GassistPi. This enables individual device and function control via the Harmony Hub. It took a long time to figure that out and in the end the response time appeared to be strangely long (for me approx. 10 seconds). But got single command TV On/Off working as proof of concept and I made use of the Harmony IR code library.