shivasiddharth/GassistPi #204: Clean Install Strange Radio Streaming Behavior

ifb777, 2018-02-11:
Installed GassistPi on a fresh raspbian install today and during testing encountered this situation:
I edited actions.py with three local stations that I know to work well.  Issued 'tune into radio one' and was greeted with silence. issued 'tune into radio two' and after an abnormaly long pause, that station started streaming.  Then tried radio three and again silence.
I Then ran Main.py in the environment.  Here is the outcome:-

ON_CONVERSATION_TURN_STARTED
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': 'tune into a radio three'}
ON_CONVERSATION_TURN_FINISHED:
  {'with_follow_on_turn': False}

### Radio Stream fails to play###

ON_CONVERSATION_TURN_STARTED
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': 'tune in to Radio 2'}
50.000000
http://www.abc.net.au/res/streaming/audio/mp3/radio_national.pls
Tuning into Radio 2
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of say.mp3 finished.
ON_CONVERSATION_TURN_FINISHED:
  {'with_follow_on_turn': False}
50.000000
2018/02/11 16:38:45 socat[4472] E connect(6, AF=1 "/tmp/mpvsocket", 16): Connection refused

**#### Radio Stream playing OK ####**

Notice that radio three is expressed as word and doesn't work yet radio 2 is expressed as a number and works as expected albeit slow to start the stream.

Anyone any ideas???

TIA

ifb777, 2018-02-11:
Further to this posting, I have just discovered that Main.py is now missing all references to Google Music Playlist, Album and Artist code.  As this is a clean install cloned earlier today, I'm wondering if there is a part of main.py missing
Input Please - I thought I was starting to understand this stuff.  Obviously not
TIA

Enzoinventor, 2018-02-11:
hello,
ON_RECOGNIZING_SPEECH_FINISHED:
{'text': 'tune into a radio three'}
this command is not correct.
the command correct is: Tuning into radio one

Enzoinventor, 2018-02-11:
sure to pronounce the command well? look in the video
https://www.youtube.com/watch?v=ae0iwJ62uaM
minute 2:17

ifb777, 2018-02-11:
> sure to pronounce the command well? look in the video 

Granted but that doesn't seem to explain why one station is expressed as a word (three) and the other as a number (2).  Mystified.

I'm still interested to know the answer to   

> Further to this posting, I have just discovered that Main.py is now missing all references to Google Music Playlist, Album and Artist code. As this is a clean install cloned earlier today, I'm wondering if there is a part of main.py missing

shivasiddharth, 2018-02-11:
1. The keyword for radio was updated to **radio** two updates back, but i forgot reflect the change in README, which have now been made. So you can say Play Radio 1 instead of the complicated tune into.

2. The three appearing as 3 and two appearing as two is the inherent behaviour of the Google's Speech to text conversion. Thats why we need to manually run the commands and then make the modifications suitably.

3. You are not missing anything in the main.py. I have just moved those parts into the actions.py file. Those lines are now under gmusicselect function in actions.py. Previously, the same piece of code was written twice, now its written once and called twice.

VinceW31, 2018-02-11:
@ifb777  I had the same issue with the number /word problem, I fixed it in actions.py as shown below and now it works fine as it detects both words and numbers:

#Radio Station links
Radio_1 = 'http://www.radiofeeds.co.uk/bbcradio1.pls'
Radio_2 = 'http://www.radiofeeds.co.uk/bbcradio2.pls'
Radio_6 = 'http://www.radiofeeds.co.uk/bbc6music.pls'
Heart = 'http://media-the.musicradio.com:80/HeartUK'

#The number of station names and station links should be the same
#and they must be in strict order relating directly to each other
stnname=('Radio one',
         'Radio on',
         'Radio 1',
         'Radio 2',
         'Radio to',
         'Radio too',
         'Radio 6',
         'Heart')#Add more stations if you want, see http://www.radiosure.com/stations/

stnlink=(Radio_1,
         Radio_1,
         Radio_1,
         Radio_2,
         Radio_2,
         Radio_2,
         Radio_6,
         Heart) # If you add more stations above, then you must add a new station link Name in this list too

Hope this helps, Enjoy!

ifb777, 2018-02-12:
Firstly many thanks to **VinceW31** for the excellent solution to Radio Station naming.  I would never have come up with that solution on my own.
To **Shivasidharth** thanks for pointing me in the right direction.  I will put all of this together this evening and give these features a good workout.

Thanks again

shivasiddharth, 2018-03-01:
Closing this for inactivity

closed 2018-03-01 by shivasiddharth