shivasiddharth/GassistPi #205: GassistPi for Orange Pi Zero

kodxana, 2018-02-12:
Is there any way to make it work on Orange Pi Zero?

shivasiddharth, 2018-02-15:
This will work on Orange pi Zero, just make sure to set  the username as 'pi'

kodxana, 2018-02-15:
Xd I just bought Rasberry Pi 3 with full kit :)

Sthopeless, 2018-05-18:
@shivasiddharth which parts/lines of RPi GPIO I need to comment out? 
I am trying to get it working on my Orange Pi PC+ with Armbian, test was working and headless mode is not working at all. perhaps you can help me?

```
pi@MagicMirror:~/GassistPi/src$ sudo systemctl status gassistpi-ok-google.service
● gassistpi-ok-google.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi-ok-google.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2018-05-18 20:34:47 CEST; 1s ago
 Main PID: 8992 (python)
   CGroup: /system.slice/gassistpi-ok-google.service
           └─8992 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id 'COPY-FROM-modelid.txt'

May 18 20:34:47 MagicMirror systemd[1]: Started Google Assistant-Voice.
pi@MagicMirror:~/GassistPi/src$ sudo systemctl status gassistpi-ok-google.service
● gassistpi-ok-google.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi-ok-google.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2018-05-18 20:35:13 CEST; 9s ago
 Main PID: 9040 (python)
   CGroup: /system.slice/gassistpi-ok-google.service
           └─9040 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id project-142249588881887248-20180518164717

May 18 20:35:13 MagicMirror systemd[1]: Started Google Assistant-Voice.

```

commit bb8c2831 (2018-11-26): Update README.md

Fixes #633 Fixes #205
closed 2018-11-26 by shivasiddharth