shivasiddharth/GassistPi #208: Crashes after chromecast commands

albzn540, 2018-02-12:
The service crashes after executing a chromecast command. Souch as "Pause chromecast".

shivasiddharth, 2018-02-12:
Make sure to assign the ipaddress of your chromecast in the chromecast function in the actions.py script.

albzn540, 2018-02-12:
It works though, the chromecast do pause the media. But the service crashes nontheless. But I'll try that!

shivasiddharth, 2018-02-12:
Attach the log, let me take a look. You can find the log in /tmp folder

albzn540, 2018-02-13:
[GassistPi.log](https://github.com/shivasiddharth/GassistPi/files/1721877/GassistPi.log)

Here's the log!
This time, I was able to say "pause chromecast" and it worked without crashing.
Saying "play chromecast" however still crashes the assistant.

shivasiddharth, 2018-02-13:
Looks like, you have not added the Youtube API key. Can you add that pls.

shivasiddharth, 2018-03-03:
Closing this for inactivity

closed 2018-03-03 by shivasiddharth