shivasiddharth/GassistPi #209: crash after update

shan2907, 2018-02-13:
all features work fine except for the media playback ones. tried streaming from youtube and playing from gmusic but the assistant just crashes . service status says
"
gassistpi-ok-google.service: Service hold-off time over, scheduling restart.

"

shivasiddharth, 2018-02-13:
Did you add the YouTube API key and Google Login Credentials again after updating ?

Rooster69, 2018-02-13:
@shivasiddharth : Hi Sid, I too have the same issue even after I have inputted the YouTube API key and Google Login credentials.

2018-02-13 15:34:37,305 DEBUG google.auth.transport.requests Making request: POST https://accounts.google.com/o/oauth2/token
2018-02-13 15:34:37,311 DEBUG requests.packages.urllib3.connectionpool Starting new HTTPS connection (1): accounts.google.com
2018-02-13 15:34:37,459 DEBUG requests.packages.urllib3.connectionpool https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2018-02-13 15:34:37,491 DEBUG requests.packages.urllib3.util.retry Converted retries value: 3 -> Retry(total=3, connect=None, read=None, redirect=None, status=None)
2018-02-13 15:34:37,503 DEBUG requests.packages.urllib3.connectionpool Starting new HTTPS connection (1): embeddedassistant.googleapis.com
2018-02-13 15:34:37,759 DEBUG requests.packages.urllib3.connectionpool https://embeddedassistant.googleapis.com:443 "GET /v1alpha2/projects/project-id/devices/D3A900393134E79D7623F9BB19FC5415 HTTP/1.1" 403 None

shan2907, 2018-02-14:
after adding the required credentials everything works fine. Mention it in the readme will help others!

Rooster69, 2018-02-14:
@shan2907 @shivasiddharth , I have added the credentials and still no luck. I have attached the log hope this help.
 
[GassistPi.log](https://github.com/shivasiddharth/GassistPi/files/1724799/GassistPi.log)

shivasiddharth, 2018-02-14:
@Rooster69 , You have entered the API-KEY without the single quotes. It should be within single quotes.

Rooster69, 2018-02-14:
Hi @shivasiddharth , Thanks for effort, there is a single quote. 
[image]

shivasiddharth, 2018-02-14:
Can you try this solution? #163

Rooster69, 2018-02-14:
It has been done prior I post. and did not help. I will try something else and will report back. Thanks for the great effort.

Keep up the great work!

Rooster69, 2018-02-14:
You would never believe this, after examining the logs carefully, there is an error "403". Which is access denied in a URL. Its happens to be that our company firewall is blocking the YouTube request.

I bought the Pi home, and all is working perfectly now. Sorry for all the trouble.

This is quite an experience as it involves the firewall.

Thank Sid for all the advices.