shivasiddharth/GassistPi #211: How to trigger Respeaker 2-mic LEDs (please, add extra event in main.py)

Pnether, 2018-02-16:
I managed to let GassistPi control the three on-board LEDs on Respeaker 2-mic on Pi3B.
The result is more fun, better interaction, and it does not require to modify GassistPi code.
Just start a python script and the LEDs will interact with GassistPi via GPIO pin 5 & 6.

**SEEED WEBPAGE TO START**
see paragraph 'How to use the on-board APA102 LEDs (controlled by SPI) on Pi 3B': 
[ http://wiki.seeed.cc/Respeaker_2_Mics_Pi_HAT/#how-to-use-the-on-board-apa102-leds) ](url)
( for more info see also [ https://github.com/respeaker/mic_hat ](url)
and [ https://www.iot-projekte.ch/apa102-led-part-1-pictures/ ](url))

**CONFIGURE Pi 3B TO USE SPI FOR RESPEAKER LED CONTROL**
`sudo raspi-config`
5 Interfacing Options  -  Configure connections to peripherals
P4 SPI                          -  Enable automatic loading 
( enable interface SPI and confirm )

**REBOOT the Pi**

**INSTALL spidev**
`git clone https://github.com/respeaker/mic_hat.git`
	`sudo pip install spidev`
	`cd mic_hat`

**TEST IF THE LEDs LIGHT UP BY RUNNING pixels.py !! MAKE THIS WORK BEFORE PROCEEDING !!**
	`python pixels.py`

I succesfully run a `/usr/bin/python` script (see below) parallel to GassistPi to control 
the Respeaker SPI LEDs. The script detects the state GPIO pin 5 and 6,
since these pins are used by GassistPi to control ordinary LEDs. (as coded in main.py).

_Note that I was not able to integrate the LED control directly in main.py 
because always an error < ImportError: No module named 'spidev' > appeared.
(with use of  `/home/pi/env/bin/python pixels.py`  the spidev module is not supported ?)_

**CREATE A NEW PYTHON SCRIPT TO DETECT THE STATE OF GPIO pins 5 and 6**
	`nano detect_GPIO_for_RespeakerLEDs.py`

**INSERT FOLLOWING LINES INTO SCRIPT detect_GPIO_for_RespeakerLEDs.py**
	
```
import RPi.GPIO as GPIO
import time

from pixels import pixels

GPIO.setmode(GPIO.BCM)

# NOTE: BELOW, THE PINS ARE DEFINED AS !!! OUT !!!
GPIO.setup(5, GPIO.OUT)
GPIO.setup(6, GPIO.OUT)
pixels.off()

BothPin56 = ''

while True:
    pin5_state = GPIO.input(5)
    pin6_state = GPIO.input(6)
    print('Pin 5 = ' + str(pin5_state) + ', Pin 6 = ' + str(pin6_state) + BothPin56
    if pin5_state == 1:
    pin6_state = GPIO.input(6)
    print('Pin 5 = ' + str(pin5_state) + ', Pin 6 = ' + str(pin6_state) + BothPin56
    if pin5_state == 1:
       pixels.think()
       time.sleep(0.6)
       BothPin56 = ''
    if pin6_state == 1:
       pixels.speak()
       time.sleep(0.6)
       BothPin56 = ''
    if pin5_state == 0 and pin6_state == 0:
       pixels.off()
       time.sleep(0.6)
       BothPin56 = ', LEDS OFF'
 
pixels.off()
GPIO.cleanup()
```

**START GassistPi IF IT IS NOT ALREADY RUNNING, AND START THE SCRIPT**
	`python detect_GPIO_for_RespeakerLEDs.py`
now, the LEDs should react upon questions and answers to/from the assistant.

**NOTE about MISSING EVENT IN main.py** 
when 'Hey Google' is spoken, but nothing is asked after that (silence only),
the LED's stay ON forever. This can be solved by adding two extra lines in main.py:

	nano main.py

**search in main.py with <ctrl W> for the handling of event 'ON_RESPONDING_FINISHED':**
    _if event.type == EventType.ON_RESPONDING_FINISHED:
       GPIO.output(6,GPIO.LOW)
       GPIO.output(5,GPIO.HIGH)
       led.ChangeDutyCycle(100)_

**add next two code lines and all LEDs are also switched OFF after 'Hey Google, and continued silence**
    `if event.type == EventType.ON_RECOGNIZING_SPEECH_FINISHED:`
       `GPIO.output(5,GPIO.LOW)`

**@shivasiddharth**
above two lines should be added anyway, also for control of normal LEDs ?

**NOTE:** It works for me, but I have only little Linux/Python/Pi knowledge, 
so I am sure a simpler, better or more elegant solution will be possible !

shivasiddharth, 2018-02-16:
> CREATE A NEW PYTHON SCRIPT TO DETECT THE STATE OF GPIO pins 5 and 6
> nano detect_GPIO_for_RespeakerLEDs.py
>
> INSERT FOLLOWING LINES INTO SCRIPT detect_GPIO_for_RespeakerLEDs.py
>
> import RPi.GPIO as [...]

All these are not required. 
Copy the apa102.py and pixels.py from [here](https://github.com/respeaker/mic_hat) and place it in the /GassistPi/src folder. 
In the main.py/pushbutton.py whatever you are using, under the list of items that i am importing, add this `from pixels import pixels`
Then comment out all my instances of GPIO5 and GPIO6, instead add `pixels.wakeup()` or `pixels.think()` or `pixels.speak()` or `pixels.listen()`

jabulonboaz, 2018-02-16:
@shivasiddharth  can you please be a bit more specific on this lat part, comment out even these? 
and if yes replace them with which of the options?
#Indicator Pins
#GPIO.setup(5, GPIO.OUT)
#GPIO.setup(6, GPIO.OUT)
#GPIO.output(5, GPIO.LOW)
#GPIO.output(6, GPIO.LOW)

Then GPIO5 and 6 apear in KODI, comment them there too? 
and what about when the LEDs are off?
if event.type == EventType.ON_CONVERSATION_TURN_TIMEOUT:
      GPIO.output(5,GPIO.LOW)
      GPIO.output(6,GPIO.LOW)

Pnether, 2018-02-16:
Thanks shivasiddharth,
Working via main.py now ! I left the GPIO commands untouched and just added the pixels commands.

After installation of spidev in the alternate python environment, the following command also worked:
`/home/pi/env/bin/python pixels.py`
and no error anymore when line 'from pixels import pixels' was added in main.py

`nano main.py`

**Insert 'from pixels import pixels', and the immediate next 2 lines to make the LEDs work:**
`from pixels import pixels`
`pixels.wakeup()`
`pixels.off()`

inserted LED command in handling of EventType.ON_CONVERSATION_TURN_STARTED:
`pixels.think()`

inserted LED command in handling of EventType.ON_RESPONDING_STARTED:
`pixels.speak()`

inserted LED command in handling of EventType.ON_RESPONDING_FINISHED:
`pixels.off()`

**EVENT ADDED TO SWITCH ALL LEDS OFF AFTER "Hey Google, and continued silence"::**
   ` if event.type == EventType.ON_RECOGNIZING_SPEECH_FINISHED:`
       `GPIO.output(5,GPIO.LOW)`
       `pixels.off()`

inserted LED command in handling of EventType.ON_CONVERSATION_TURN_FINISHED:
`pixels.off()`

jabulonboaz, 2018-02-16:
@Pnether any chance to please share your main.py so i can figure out what to change what exactly please?

shivasiddharth, 2018-02-16:
@Pnether , There Respeaker 2-mic has no overlapping pins with my existing declarations, so there is nor need to comment anything. 
Just a small tip, next time when you  get `ImportError: No module named ` anything it could be two things, either the modules or not installed within the environment or the script is being run outside the environment. To doublecheck your package installation, open the env folder in /home/pi directory and then into /lib/Pyhton/site-packages/ and there you can find all your installed packages.

Pnether, 2018-02-17:
@jabulonboaz: 
maybe better to start with an original main.py because that code indeed  is critical.
Don't know how to send yoy my main.py (whis is already a mass experiments). I located 'from pixels import pixels' after the line 'with from actions import chromecast_control'..
For the rest of the LED commands, the structure of all events handled looks the same is in the lines for the added event EventType.ON_RECOGNIZING_SPEECH_FINISHED. That added example event contains all the necessary lines to control both normal LED via GPIO and the Respeaker LEDs.
@shivasiddharth 
I already became aware of the different python environments. Thans for explaining how to handle that.. When controlled via main.py the LEDs control is more direct and has better response time.

jabulonboaz, 2018-02-18:
@Pnether  I followed what u wrote to the letter can you please at least just copy paste these 3 or 4 places where you added anything to do with the led control including say the lone you added and the line before it and after it?

jabulonboaz, 2018-02-18:
I did it! :) i activated SPI and it started working thank you @Pnether and @shivasiddharth 

@shivasiddharth  2 observations for others at are a beginner level like me
You mention:   
"All these are not required."   ---  The activation of SPI was required though from Panther's description
"source env/bin/activate/"  should be "source env/bin/activate"

I hope you integrate all of this to your build the LEDs of Respeker 2mic look great ! I wish we can have the button integration fo the respeaker too but without it crashing Alexa on AssistantsPi (this is what i am using).

shivasiddharth, 2018-02-18:
Closing this issue then !

closed 2018-02-18 by shivasiddharth

Pnether, 2018-02-18:
@jabulonboaz: 
glad you succeeded. I also enjoy to watch the LEDs patterns very much and it improves interaction !

@shivasiddharth: 
To trap correct recognition and program flow at this moment, I split the `say()` confirmation in two parts. The first part of confirmation is in main.py with `say('Ok')` and the second part in actions.py, like `say('Shutting down Raspberry Pi')`. Then I know immediately which voice part is not recognised correctly and which script should be modified.

Now, also LED confirmation can be used as an alternative for this. Good idea. I guess that anybody who is interested can add such LED confirmations, since the principle to light up the LEDs is known right now.
Thanks again shivasiddharth !

jabulonboaz, 2018-02-27:
@pnether and @Shivasiddharth 
Is there a way to triger the Respeaker LEDs for Alexa on Assitants pi? i like to do a smilar thing like you did for GA perhaps with blue colour for Alexa.

Pnether, 2018-02-27:
@jabulonboaz: Sorry, can not help, since I do not use Alexa

jabulonboaz, 2018-02-27:
@Pnether thank you any way.