shivasiddharth/GassistPi #212: [Question] Can't modify "actions.py" to invoke a shell script via GA voice command - what am i doing wrong?

jabulonboaz, 2018-02-16:
@shivasiddharth and all why when i amend the "actions.py" file like this (bellow) GA does not startup at boot any more? when i comment out all my amendments it works again.
I have a script that switches on and off the WiFi of my router that works when i execute it from a terminal 
 in the Pi3 that i have GassistPi running and i am trying to invoke it via a voice command similar to the one you created for "shutdown" .  e.g. "Hey Google trigger internet" and then sudo ./wifi5ghz_off.sh is executed.
I have been trying many different combinations for some time to get it to work but i am no coder i am afraid.  if anyone can please help me unstuck i will be graceful. 

#GPIO Device Control
def Action(phrase):
    if 'internet' in phrase:
        say("Turning off Wi-Fi")
	time.sleep(10)
        os.system("sudo /home/pi/wifi5ghz_off.sh") 
        #os.system("sudo ./wifi5ghz_off.sh")
       #subprocess.call(["sudo ./wifi5ghz_off.sh"], shell=True, stdout=subprocess.PIPE, stderr=subprocess.PIPE)
    if 'shutdown' in phrase:
        say("Shutting down Raspberry Pi")
        time.sleep(10)
        os.system("sudo shutdown -h now")
        #subprocess.call(["shutdown -h now"], shell=True, stdout=subprocess.PIPE, stderr=subprocess.PIPE)

Enzoinventor, 2018-02-17:
all the changes you create in action.py must also be called in main.py

jabulonboaz, 2018-02-18:
There was no need to change anything in main.py unless you want to change the trigger word but I left it as it was. I found the solution and it was really very simple I am surprised nobody pointed it out to me before. I removed side and it worked this is an example of the code that I inserted near the end of the actions.py file near the ‘trigger shutdown’ part. 

if 'script' in phrase:
    say("Initiating script")
    os.system("./my script.sh")

And this is how you can voice trigger run a shell script file locally in GassistPi and I bet it works with GA AYI too

closed 2018-02-18 by jabulonboaz