shivasiddharth/GassistPi #213: information

aandroide, 2018-02-17:
hello I wanted 2 information:
1 how to stop youtube when a song ends? (I do not want the replay of the song I asked)
2 every now and then google emits the sound that is listening even without saying the word ok google this happens especially if the tv is on.
It would be useful to insert a timer that if nothing is pronounced within 30 seconds, the LED goes off instead of blinking at infinity.
Thanks so much

Pnether, 2018-02-18:
1. Do you refer the `stream` keyword ? I Do not recognise that problem: when I use the voice command `Hey Google, stream <song>` then the music stops after the song has ended.
2. I experience the same when watching TV and also sometimes wonder why the assistant is triggered. I think that correct voice recognition is still in its infancy.
3. Also for me the LEDs did not switch off in that situation and solved that by adding an extra event in main.py, as below:

**search in main.py with ctrl-W for the handling of event 'ON_RESPONDING_FINISHED':**
if event.type == EventType.ON_RESPONDING_FINISHED:
GPIO.output(6,GPIO.LOW)
GPIO.output(5,GPIO.HIGH)
led.ChangeDutyCycle(100)

**add next two code lines and all LEDs are also switched OFF after 'Hey Google, and continued silence**
if event.type == EventType.ON_RECOGNIZING_SPEECH_FINISHED:
GPIO.output(5,GPIO.LOW)

Enzoinventor, 2018-02-18:
@Pnether 
Hello,
yes, I refer to the word Hey Google, stream <song> and when the song ends, after 5 seconds it starts again from the beginning the requested song as if a loop was inserted automatically.
I tried to insert the suggested line in main.py but when I say google hei and I do not talk he flashes endlessly. this is what is to be corrected. after 30 sec of silence it should stop blinking.

Enzoinventor, 2018-02-18:
I just fixed the LED that always blinks when you do not speak, adding these lines to your code:
if event.type == EventType.ON_RECOGNIZING_SPEECH_FINISHED:
       GPIO.output (5, GPIO.LOW)
       GPIO.output (6, GPIO.LOW)
       led.ChangeDutyCycle (0)
after about 8 seconds of silence it turns off.
only the youtube replay problem remains

aandroide, 2018-02-18:
@Pnether
Hello,
yes, I refer to the word Hey Google, stream and when the song ends, after 5 seconds it starts again from the beginning the requested song as if a loop was inserted automatically.

I just fixed the LED that always blinks when you do not speak, adding these lines to your code:
if event.type == EventType.ON_RECOGNIZING_SPEECH_FINISHED:
GPIO.output (5, GPIO.LOW)
GPIO.output (6, GPIO.LOW)
led.ChangeDutyCycle (0)
after about 8 seconds of silence it turns off.
only the youtube replay problem remains

Pnether, 2018-02-18:
Well done, and thanks for the feedback! Since I use the built-on APA102 LED on a Respeaker 2-mic hat (which are SPI controlled), I do not use GPIO controlled LEDS. So could not really test it. Glad you solved it !
But I can not help with the YouTube  replay, since I do not have that problem.

aandroide, 2018-02-24:
hello,
I think the error that I find the loop continuous even if I say stream instead of autoplay is in this part of code:
should the zero value not be declared?
def YouTube_Autoplay(phrase):
    urllist=[]
    idx=phrase.find('stream')
    track=phrase[idx:]
    track=track.replace("'}", "",1)
    track = track.replace('stream','',1)
    track=track.strip()
    say("Getting autoplay links")
    fullurl,urlid=youtube_search(track)
    autourls=fetchautoplaylist(fullurl,10)#Maximum of 10 URLS
    print(autourls)
    for i in range(0,len(autourls)):
        urllist.append(autourls[i])
    say("Adding autoplay links to the playlist")
    with open('/home/pi/youtubeurllist.json', 'w') as output_file:
        json.dump(autourls, output_file)
    if os.path.isfile("/home/pi/.youtubeplayer.json"):
        os.remove('/home/pi/.youtubeplayer.json')
    youtubeplayer()

def YouTube_No_Autoplay(phrase):
    urllist=[]
    idx=phrase.find('stream')
    track=phrase[idx:]
    track=track.replace("'}", "",1)
    track = track.replace('stream','',1)
    track=track.strip()
    say("Getting youtube link")
    fullurl,urlid=youtube_search(track)
    urllist.append(fullurl)
    print(urllist)
    with open('/home/pi/youtubeurllist.json', 'w') as output_file:
        json.dump(urllist, output_file)
    if os.path.isfile("/home/pi/.youtubeplayer.json"):
        os.remove('/home/pi/.youtubeplayer.json')
    youtubeplayer()

shivasiddharth, 2018-02-24:
@aandroide , #230 will fix that, if you want to test the fix, copy paste the stream command part from the [developmental branch](https://github.com/shivasiddharth/GassistPi/tree/develop-1) main.py into your existing script and check.

aandroide, 2018-02-24:
@shivasiddharth  Hi Sid, well done now works

closed 2018-02-24 by shivasiddharth