shivasiddharth/GassistPi #217: Bluealsa transport: no such device

legemulo, 2018-02-18:
Hello! This is a fresh install with bluetooth speakers. I can hear the assistant but I get the following error:
ALSA lib bluealsa-pcm.c:659:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA transport: No such device
[1034:1062:ERROR:audio_input_processor.cc(748)] Input error

The device has a built-in microphone. [here](https://www.ebay.com/itm/Enusic-Soundcup-Bluetooth-Outdoor-Speaker-With-EQ-Mode-Waterproof-NFC-20w-/112301764665)
I also have a mini usb microphone which worked well in the previous install.
Thanks in advance.

shivasiddharth, 2018-02-19:
Please post the outputs of `aplay -l `and `arecord -l`.

legemulo, 2018-02-20:
At first, the output of arecord -l said there was no sound card and aplay sent a harsh noise to the speakers. Then I connected the device to the jack output and finished the installation of Google Assistant. It is working like a champ now, but no bluetooth connection joy.
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0

the device is not connected as a bluetooth one but it shows
pi@raspberrypi:~ $ busctl tree org.bluez
└─/org
  └─/org/bluez
    └─/org/bluez/hci0
      └─/org/bluez/hci0/dev_FC_58_FA_23_95_10
pi@raspberrypi:~ $ sudo systemctl status bluet*
● bluetooth.target - Bluetooth
   Loaded: loaded (/lib/systemd/system/bluetooth.target; static; vendor preset: enabled)
   Active: active since Mon 2018-02-19 14:01:00 -03; 6h ago
     Docs: man:systemd.special(7)

Feb 19 14:01:00 raspberrypi systemd[1]: Reached target Bluetooth.

● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2018-02-19 14:01:00 -03; 6h ago
     Docs: man:bluetoothd(8)
 Main PID: 528 (bluetoothd)
   Status: "Running"
   CGroup: /system.slice/bluetooth.service
           └─528 /usr/lib/bluetooth/bluetoothd

Feb 19 14:01:00 raspberrypi systemd[1]: Starting Bluetooth service...
Feb 19 14:01:00 raspberrypi bluetoothd[528]: Bluetooth daemon 5.43
Feb 19 14:01:00 raspberrypi systemd[1]: Started Bluetooth service.
Feb 19 14:01:00 raspberrypi bluetoothd[528]: Starting SDP server
Feb 19 14:01:00 raspberrypi bluetoothd[528]: Bluetooth management interface 1.14 initialized
Feb 19 14:01:00 raspberrypi bluetoothd[528]: Failed to obtain handles for "Service Changed" characteristic
Feb 19 14:01:00 raspberrypi bluetoothd[528]: Sap driver initialization failed.
Feb 19 14:01:00 raspberrypi bluetoothd[528]: sap-server: Operation not permitted (1)
Feb 19 14:01:00 raspberrypi bluetoothd[528]: Endpoint registered: sender=:1.6 path=/A2DP/SBC/Source/1
Feb 19 14:01:00 raspberrypi bluetoothd[528]: Endpoint registered: sender=:1.6 path=/A2DP/SBC/Sink/1

 With the device connected as bluetooth, the assistant stopped answering.
pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

pi@raspberrypi:~ $ busctl tree org.bluez
└─/org
  └─/org/bluez
    └─/org/bluez/hci0
      └─/org/bluez/hci0/dev_FC_58_FA_23_95_10
        └─/org/bluez/hci0/dev_FC_58_FA_23_95_10/fd0

pi@raspberrypi:~ $ sudo systemctl status bluet*
● bluetooth.target - Bluetooth
   Loaded: loaded (/lib/systemd/system/bluetooth.target; static; vendor preset: enabled)
   Active: active since Mon 2018-02-19 21:01:09 -03; 5min ago
     Docs: man:systemd.special(7)

Feb 19 21:01:09 raspberrypi systemd[1]: Reached target Bluetooth.

● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2018-02-19 21:01:09 -03; 5min ago
     Docs: man:bluetoothd(8)
 Main PID: 528 (bluetoothd)
   Status: "Running"
   CGroup: /system.slice/bluetooth.service
           └─528 /usr/lib/bluetooth/bluetoothd

Feb 19 21:01:09 raspberrypi bluetoothd[528]: Bluetooth daemon 5.43
Feb 19 21:01:09 raspberrypi systemd[1]: Started Bluetooth service.
Feb 19 21:01:09 raspberrypi bluetoothd[528]: Starting SDP server
Feb 19 21:01:09 raspberrypi bluetoothd[528]: Bluetooth management interface 1.14 initialized
Feb 19 21:01:09 raspberrypi bluetoothd[528]: Failed to obtain handles for "Service Changed" characteristic
Feb 19 21:01:09 raspberrypi bluetoothd[528]: Sap driver initialization failed.
Feb 19 21:01:09 raspberrypi bluetoothd[528]: sap-server: Operation not permitted (1)
Feb 19 21:01:09 raspberrypi bluetoothd[528]: Endpoint registered: sender=:1.9 path=/A2DP/SBC/Source/1
Feb 19 21:01:10 raspberrypi bluetoothd[528]: Endpoint registered: sender=:1.9 path=/A2DP/SBC/Sink/1
Feb 19 21:03:10 raspberrypi bluetoothd[528]: Endpoint registered: sender=:1.9 path=/A2DP/SBC/Source/2

I don't know much about it all, but it's not hard to guess that the sound files have to have the info about the device being connected as line in and bluetooth for it to work.

shivasiddharth, 2018-02-20:
Can you check this out: https://youness.net/raspberry-pi/how-to-connect-bluetooth-headset-or-speaker-to-raspberry-pi-3

shivasiddharth, 2018-03-15:
Closing this for inactivity

closed 2018-03-15 by shivasiddharth