shivasiddharth/GassistPi #218: Google Assistant restarts when I try to stream from Google Music

jbhatelia, 2018-02-19:
I have entered my email address and its password in the actions.py file

When ever i say `"Hey Google,Play songs by artist <ARTIST NAME> from Google Music "` the Google assistant restarts..

so to check the issue I manually started assistant with 
`/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id 'my project id'  --device_model_id 'mymodel id'`
and this is what i got -

```
ON_START_FINISHED

ON_CONVERSATION_TURN_STARTED
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': 'play songs by artist Arijit Singh on Google Music'}
**sh: 1: cannot create /home/pi/GassistPi/src/trackchange.py: Permission denied**
arijit singh
pi@raspberrypi:~ $ 
```

shivasiddharth, 2018-02-19:
This is an issue with raspbian. The user has not been granted root access. There is a ton of info to fix that : https://www.google.ie/search?q=granting+root+access+to+user+raspbian&oe=UTF-8

parthsuthar, 2018-03-12:
Tried this as well. Gave root privileges to 'pi' user - which was supposedly a blocker to access gmusic. Doesn't work. Any further help ?

shivasiddharth, 2018-03-15:
Please attach the Log file here. You can find the log file in /tmp directory

parthsuthar, 2018-03-18:
Manual log-
ON_CONVERSATION_TURN_STARTED
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': "what's the weather"}
ON_RESPONDING_STARTED:
  {'is_error_response': False}
ON_RESPONDING_FINISHED
ON_CONVERSATION_TURN_FINISHED:
  {'with_follow_on_turn': False}

ON_CONVERSATION_TURN_STARTED
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': 'play songs by Arijit Singh from Google Music'}
ON_CONVERSATION_TURN_FINISHED:
  {'with_follow_on_turn': False}

/Gassist Log -
2018-03-18 15:25:58,598 DEBUG google.auth.transport.requests Making request: POST https://accounts.google.com/o/oauth2/token
2018-03-18 15:25:58,609 DEBUG requests.packages.urllib3.connectionpool Starting new HTTPS connection (1): accounts.google.com
2018-03-18 15:25:59,066 DEBUG requests.packages.urllib3.connectionpool https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2018-03-18 15:27:24,760 DEBUG google.auth.transport.requests Making request: POST https://accounts.google.com/o/oauth2/token
2018-03-18 15:27:24,777 DEBUG requests.packages.urllib3.connectionpool Starting new HTTPS connection (1): accounts.google.com
2018-03-18 15:27:25,140 DEBUG requests.packages.urllib3.connectionpool https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None

parthsuthar, 2018-03-18:
I can hear the weather but there isn't any output on Google music query

shivasiddharth, 2018-03-18:
Without looking at the full log file, i cant say much

parthsuthar, 2018-03-18:
This is the full one I got. Tellme specifically and I can publish the log

shivasiddharth, 2018-03-18:
Have you added your google login credentials in the actions.py script ? If you are using two step authentication, you need to enter app specific password. Also, You need to have music uploaded in play music cloud. And after adding music to the cloud you need to refresh the music list using: `Hey Google, Refresh Music`. Sometimes the music volume might be low, so try changing the media volume (specifics are given in the readme doc).
What you have added is not the entire log file.

parthsuthar, 2018-03-18:
1. Yes, added the app specific password
2. Uploaded Music - why so ? It should be able to search and play from Google Music (online). E.g. such as Top Charts (which is Playlist)
3. Checked Volume > OK
4. Full log file > This is the final output in the log file (in /tmp). Do I need to look somewhere else ?

parthsuthar, 2018-03-18:
how do i check/debug the process of gmusic api is logging in (with the credentials provided) and parsing the song requests ? since its parsing other requests for time/date etc.

shivasiddharth, 2018-03-18:
`Uploaded Music - why so ? It should be able to search and play from Google Music (online). E.g. such as Top Charts (which is Playlist). I have two albums added as well in the Music Library`
That is how the API works For more details on the API please go through their official (repo) [https://github.com/simon-weber/gmusicapi].
If you have added the music, then you should refresh the local music list as i indicated previously. In any case you should get a voice prompt giving a reason as to why music is not playing, like "No matching albums found" or "No Matching artist found" etc.
Frankly, this is not the full log file. This is just a part of the file that you have posted. I am sorry, but cant help any further without detailed info.

parthsuthar, 2018-03-18:
@shivasiddharth I have uploaded whatever output is there in the log file. why do you think I have any reason to not upload the full log (when I am really looking forward to trouble shoot this problem) ?

Tellme any specific command which I need to enter to get the full log file you are mentioning. Currently, this is the only output I have received in the log file!

parthsuthar, 2018-03-18:
@shivasiddharth Tried adding Music. Created Playlists. When trying to play a Playlist > GA says "playing song from your Playlist" then restarts

shivasiddharth, 2018-03-18:
Did you refresh the music list using: `"Hey google, refresh music list"` command ?

parthsuthar, 2018-03-18:
yes, so after double efforts - Play All songs from Google Music works (but it distorts/stutters a lot)

parthsuthar, 2018-03-18:
All other commands like play Playlist etc isn't working

parthsuthar, 2018-03-18:
ok, i think its getting there - its limited functionality (since it can't search & play) but should be able to get the basic commands working in due time. any ideas why is it sorting (even at low volume) ?

shivasiddharth, 2018-03-18:
Distortion can be a hardware/network issue. Since the primary issue has been resolved closing this thread. Feel free to reopen just in case.

closed 2018-03-18 by shivasiddharth

parthsuthar, 2018-03-18:
cool. thanks @shivasiddharth