shivasiddharth/GassistPi #219: How to access other library from main.py ?

dwh6411, 2018-02-19:
Hi 
Thanks for the job, it works quite well on my Pi2. I need adapt it to my use now.
I have problem to access general library from Main.py or actions.py
for example : I install LCD module Luma.LCD  with  "sudo -H pip install --upgrade luma.lcd"
and I can run my LCD outside , However I  import luma.lcd  inside  main.py, it cannot find the module.
It is the same that cannot find urllib2 , which can be run outside of main.py.
I find that : gassistant use its own python  from " /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py .."  
which python  get result ==> /usr/bin/python  
My question is that :
How can I make  general library accessible from  main.py  with /home/pi/env/bin/python  ?
Do I need  reinstall the module with pip3 from  /home/pi/env/bin/ ? or just pip  ( which pip)?  
Thanks !
weihua

shivasiddharth, 2018-02-19:
You need to install the packages within environment.
First move into the environment using:
`source env/bin/activate`
Then use pip commands to install. Just pip. We are currently working in Python3 environment and hence i dont think urllib2 installation is possible. But as a part of the installation we are already installing urllib3. So its just a matter of finding the components. A quick google search can help you spot urllib2 equivalents in ulrlib3 (subscript 2 and 3 and not interchangeable as the components of the packages are not bundled similarly).

dwh6411, 2018-02-20:
Hi shivasiddharth,
Thank you for your guide, I will check with that.

dwh6411, 2018-02-20:
Hi shivasiddharth,
I source the environment as you mentioned and reinstall with pip 
==>
pi@Pi2:~ $ source env/bin/activate
(env) pi@Pi2:~ $  sudo -H pip install --upgrade luma.lcd
Requirement already up-to-date: luma.lcd in /usr/local/lib/python2.7/dist-packages
Requirement already up-to-date: luma.core>=1.1.1 in /usr/local/lib/python2.7/dist-packages                  (from luma.lcd)
Requirement already up-to-date: pillow>=4.0.0 in /usr/local/lib/python2.7/dist-packages (f                 rom luma.core>=1.1.1->luma.lcd)
Requirement already up-to-date: monotonic; python_version < "3.3" in /usr/local/lib/python                 2.7/dist-packages (from luma.core>=1.1.1->luma.lcd)
Requirement already up-to-date: smbus2 in /usr/local/lib/python2.7/dist-packages (from lum                 a.core>=1.1.1->luma.lcd)
Requirement already up-to-date: RPI.GPIO; platform_system == "Linux" in /usr/lib/python2.7                 /dist-packages (from luma.core>=1.1.1->luma.lcd)
Requirement already up-to-date: spidev; platform_system == "Linux" in /usr/lib/python2.7/d                 ist-packages (from luma.core>=1.1.1->luma.lcd)
=====
if I import Luma module:
from luma.core.interface.serial import spi
I still  get the  error : 
ImportError: No module named 'luma'

you can see, the pip  only used with python 2.7  However Gassistant using Python 3.5 ?  
Any ideas ?
By the way, I find my urllib3 equivalent api. 
Thanks.

dwh6411, 2018-02-20:
Hi shivasiddharth,
I source the environment as you mentioned and reinstall with pip 
==>
pi@Pi2:~ $ source env/bin/activate
(env) pi@Pi2:~ $  sudo -H pip install --upgrade luma.lcd

Requirement already up-to-date: luma.lcd in /usr/local/lib/python2.7/dist-packages
Requirement already up-to-date: luma.core>=1.1.1 in /usr/local/lib/python2.7/dist-packages (from luma.lcd)
Requirement already up-to-date: pillow>=4.0.0 in /usr/local/lib/python2.7/dist-packages (from luma.core>=1.1.1->luma.lcd)
Requirement already up-to-date: monotonic; python_version < "3.3" in /usr/local/lib/python2.7/dist-packages (from luma.core>=1.1.1->luma.lcd)
Requirement already up-to-date: smbus2 in /usr/local/lib/python2.7/dist-packages (from luma.core>=1.1.1->luma.lcd)
Requirement already up-to-date: RPI.GPIO; platform_system == "Linux" in /usr/lib/python2.7/dist-packages (from luma.core>=1.1.1->luma.lcd)
Requirement already up-to-date: spidev; platform_system == "Linux" in /usr/lib/python2.7/dist-packages (from luma.core>=1.1.1->luma.lcd)

=====
if I import Luma module in main.py:
from luma.core.interface.serial import spi
I still  get the  error : 
ImportError: No module named 'luma'

As you can see, the pip  only used with python 2.7  However Gassistant using Python 3.5 ?  
Any ideas ?
By the way, I find my urllib3 equivalent api. 
Thanks.

shivasiddharth, 2018-02-20:
When you use a `sudo` command, the installation happens in the root directory. Command should be just `pip install`

dwh6411, 2018-02-20:
Hi shivasiddharth,
Thanks,   Now the error disappear ! 
👍

shivasiddharth, 2018-02-20:
Ok, closing this issue then !

closed 2018-02-20 by shivasiddharth