shivasiddharth/GassistPi #220: Start systemctl service Crash

Mandarin974, 2018-02-20:
Hi !
I followed the step-by-step instructions and everything works.
Except when I enter the "sudo systemctl start gassistpi-ok-google. service" command.
The sound starts, stops and repeats.
I don't know where his logs would allow me to find the cause of crashes.
Could you help me find his log files and find out where my error is located?

shivasiddharth, 2018-02-20:
Most likely, you would forgotten to edit the ids in the service file or the ids could be wrong. But anyways, the log file will be in /tmp/ folder

Rooster69, 2018-02-21:
@Mandarin974 : It seem that you have enabled both the  "gassistpi-push-button.service" and  "gassistpi-ok-google.service".

Disable the  "gassistpi-push-button.service" and and it should work properly.

Mandarin974, 2018-03-03:
Hello, Thank you for your help, it was indeed an ID problem,
I didn't understand that the id had to be retrieved at the time of installing the wizard, just after entering the verification code through the given link.
Thanks again for your help and see you soon!

shivasiddharth, 2018-03-06:
Closing this issue !

closed 2018-03-06 by shivasiddharth