shivasiddharth/GassistPi #221: kodi can't Youtube streaming

nice824524519, 2018-02-21:
GassistPi.log

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 37, in <module>
    from oauth2client.locked_file import LockedFile
ImportError: No module named 'oauth2client.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/__init__.py", line 41, in autodetect
    from . import file_cache
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 41, in <module>
    'file_cache is unavailable when using oauth2client >= 4.0.0')
ImportError: file_cache is unavailable when using oauth2client >= 4.0.0
2018-02-20 13:31:44,148 INFO googleapiclient.discovery URL being requested: GET https://www.googleapis.com/discovery/v1/apis/youtube/v3/rest
2018-02-20 13:31:44,516 INFO googleapiclient.discovery URL being requested: GET https://www.googleapis.com/youtube/v3/search?alt=json&key=AIzaSyDJjFw5VYMtnzDk_1aJBJaQRwql-C9TQos&part=id%2Csnippet&q=step+add-
2018-02-20 13:31:45,194 DEBUG requests.packages.urllib3.connectionpool Starting new HTTP connection (1): 192.168.3.138
2018-02-20 13:31:58,806 DEBUG requests.packages.urllib3.connectionpool http://192.168.3.138:8080 "POST /jsonrpc HTTP/1.1" 200 38
2018-02-20 13:31:58,924 DEBUG requests.packages.urllib3.connectionpool https://translate.google.com:443 "GET /translate_a/single?hl=en&client=t&tk=234563.386078&sl=auto&oe=UTF-8&dt=at&dt=bd&dt=ex&dt=ld&dt=md&dt=qca&dt=rw&dt=rm&dt=ss&dt=t&q=Playing+YouTube+video+on+Kodi&tl=en&ssel=0&ie=UTF-8&tsel=0&otf=1 HTTP/1.1" 200 None
2018-02-20 13:31:58,943 DEBUG requests.packages.urllib3.connectionpool Starting new HTTPS connection (1): translate.google.com
2018-02-20 13:31:59,257 DEBUG requests.packages.urllib3.connectionpool https://translate.google.com:443 "GET / HTTP/1.1" 200 None
2018-02-20 13:31:59,280 DEBUG requests.packages.urllib3.connectionpool Starting new HTTPS connection (1): translate.google.com
2018-02-20 13:31:59,625 DEBUG requests.packages.urllib3.connectionpool https://translate.google.com:443 "GET /translate_tts?tk=234563.386078&tl=en&client=tw-ob&total=1&ttsspeed=1&ie=UTF-8&textlen=29&q=Playing+YouTube+video+on+Kodi&idx=0 HTTP/1.1" 200 11040
2018-02-20 13:32:31,953 DEBUG requests.packages.urllib3.connectionpool Starting new HTTP connection (1): 192.168.3.138
2018-02-20 13:32:31,963 DEBUG requests.packages.urllib3.connectionpool http://192.168.3.138:8080 "POST /jsonrpc HTTP/1.1" 200 65
2018-02-20 13:32:31,984 DEBUG requests.packages.urllib3.connectionpool Starting new HTTP connection (1): 192.168.3.138
2018-02-20 13:32:32,220 DEBUG requests.packages.urllib3.connectionpool http://192.168.3.138:8080 "POST /jsonrpc HTTP/1.1" 200 38

nice824524519, 2018-02-21:
kodi.log

12:58:17.372 T:1699476224  NOTICE: [plugin.video.youtube] Running: YouTube (5.5.0) on Krypton (Kodi-17.6) with Python 2.7.13
12:58:17.397 T:1699476224 WARNING: [plugin.video.youtube] DEPRECATED "plugin://plugin.video.youtube/?action=play_video&videoid=gIa3ftX20yo"
12:58:17.397 T:1699476224 WARNING: [plugin.video.youtube] USE INSTEAD "plugin://plugin.video.youtube/play/?video_id=gIa3ftX20yo"
12:58:19.546 T:1699476224   ERROR: EXCEPTION Thrown (PythonToCppException) : -->Python callback/script returned the following error<--
                                             - NOTE: IGNORING THIS CAN LEAD TO MEMORY LEAKS!
                                            Error Type: <type 'exceptions.Exception'>
                                            Error Contents: Signature function not found
                                            Traceback (most recent call last):
                                              File "/home/osmc/.kodi/addons/plugin.video.youtube/resources/lib/default.py", line 7, in <module>
                                                runner.run(__provider__)
                                              File "/home/osmc/.kodi/addons/plugin.video.youtube/resources/lib/youtube_plugin/kodion/runner.py", line 40, in run
                                                __RUNNER__.run(provider, context)
                                              File "/home/osmc/.kodi/addons/plugin.video.youtube/resources/lib/youtube_plugin/kodion/impl/xbmc/xbmc_runner.py", line 24, in run
                                                results = provider.navigate(context)
                                              File "/home/osmc/.kodi/addons/plugin.video.youtube/resources/lib/youtube_plugin/kodion/abstract_provider.py", line 87, in navigate
                                                result = method(context, re_match)
                                              File "/home/osmc/.kodi/addons/plugin.video.youtube/resources/lib/youtube_plugin/kodion/abstract_provider.py", line 118, in _internal_root
                                                return self.on_root(context, re_match)
                                              File "/home/osmc/.kodi/addons/plugin.video.youtube/resources/lib/youtube_plugin/youtube/provider.py", line 825, in on_root
                                                return yt_old_actions.process_old_action(self, context, re_match)
                                              File "/home/osmc/.kodi/addons/plugin.video.youtube/resources/lib/youtube_plugin/youtube/helper/yt_old_actions.py", line 53, in process_old_action
                                                return _process_play_video(provider, context, re_match)
                                              File "/home/osmc/.kodi/addons/plugin.video.youtube/resources/lib/youtube_plugin/youtube/helper/yt_old_actions.py", line 19, in _process_play_video
                                                return provider.on_play(new_context, re_match)
                                              File "/home/osmc/.kodi/addons/plugin.video.youtube/resources/lib/youtube_plugin/kodion/register_provider_path.py", line 11, in wrapper
                                                return func(*args, **kwargs)
                                              File "/home/osmc/.kodi/addons/plugin.video.youtube/resources/lib/youtube_plugin/youtube/provider.py", line 475, in on_play
                                                return yt_play.play_video(self, context, re_match)
                                              File "/home/osmc/.kodi/addons/plugin.video.youtube/resources/lib/youtube_plugin/youtube/helper/yt_play.py", line 19, in play_video
                                                video_streams = client.get_video_streams(context, video_id)
                                              File "/home/osmc/.kodi/addons/plugin.video.youtube/resources/lib/youtube_plugin/youtube/client/youtube.py", line 79, in get_video_streams
                                                video_streams = video_info.load_stream_infos(video_id, player_config, cookies)
                                              File "/home/osmc/.kodi/addons/plugin.video.youtube/resources/lib/youtube_plugin/youtube/helper/video_info.py", line 403, in load_stream_infos
                                                return self._method_get_video_info(video_id, player_config, cookies)
                                              File "/home/osmc/.kodi/addons/plugin.video.youtube/resources/lib/youtube_plugin/youtube/helper/video_info.py", line 738, in _method_get_video_info
                                                parse_to_stream_list(url_encoded_fmt_stream_map)
                                              File "/home/osmc/.kodi/addons/plugin.video.youtube/resources/lib/youtube_plugin/youtube/helper/video_info.py", line 700, in parse_to_stream_list
                                                url += '&signature=%s' % cipher.get_signature(stream_map['s'])
                                              File "/home/osmc/.kodi/addons/plugin.video.youtube/resources/lib/youtube_plugin/youtube/helper/signature/cipher.py", line 24, in get_signature
                                                json_script = function_cache.get(FunctionCache.ONE_DAY, self._load_json_script, self._java_script_url)
[... cut ...]
                                                return self._load_java_script(java_script)
                                              File "/home/osmc/.kodi/addons/plugin.video.youtube/resources/lib/youtube_plugin/youtube/helper/signature/cipher.py", line 62, in _load_java_script
                                                raise Exception('Signature function not found')
                                            Exception: Signature function not found
                                            -->End of Python script error report<--

shivasiddharth, 2018-02-21:
You can find the answer in lines 4 and 5 of the Kodi log. 
Open the actions.py file and change line 338 from
`kodi.Player.open(item={"file":"plugin://plugin.video.youtube/?action=play_video&videoid=" + urlid})`
to 
`kodi.Player.open(item={"file":"plugin://plugin.video.youtube/play/?video_id=" + urlid})`

Let me know if it worked, will change it in the git master.

nice824524519, 2018-02-21:
not work

kodi.log 

NOTICE: [plugin.video.youtube] Running: YouTube (5.5.0) on Krypton (Kodi-17.6) with Python 2.7.13
16:36:23.574 T:1707078400   ERROR: EXCEPTION Thrown (PythonToCppException) : -->Python callback/script returned the following error<--
                                             - NOTE: IGNORING THIS CAN LEAD TO MEMORY LEAKS!
                                            Error Type: <type 'exceptions.Exception'>
                                            Error Contents: Signature function not found
                                            Traceback (most recent call last):
                                              File "/home/osmc/.kodi/addons/plugin.video.youtube/resources/lib/default.py", line 7, in <module>
                                                runner.run(__provider__)
                                              File "/home/osmc/.kodi/addons/plugin.video.youtube/resources/lib/youtube_plugin/kodion/runner.py", line 40, in run
                                                __RUNNER__.run(provider, context)
                                              File "/home/osmc/.kodi/addons/plugin.video.youtube/resources/lib/youtube_plugin/kodion/impl/xbmc/xbmc_runner.py", line 24, in run
                                                results = provider.navigate(context)
                                              File "/home/osmc/.kodi/addons/plugin.video.youtube/resources/lib/youtube_plugin/kodion/abstract_provider.py", line 87, in navigate
                                                result = method(context, re_match)
                                              File "/home/osmc/.kodi/addons/plugin.video.youtube/resources/lib/youtube_plugin/kodion/register_provider_path.py", line 11, in wrapper
                                                return func(*args, **kwargs)
                                              File "/home/osmc/.kodi/addons/plugin.video.youtube/resources/lib/youtube_plugin/youtube/provider.py", line 475, in on_play
                                                return yt_play.play_video(self, context, re_match)
                                              File "/home/osmc/.kodi/addons/plugin.video.youtube/resources/lib/youtube_plugin/youtube/helper/yt_play.py", line 19, in play_video
                                                video_streams = client.get_video_streams(context, video_id)
                                              File "/home/osmc/.kodi/addons/plugin.video.youtube/resources/lib/youtube_plugin/youtube/client/youtube.py", line 79, in get_video_streams
                                                video_streams = video_info.load_stream_infos(video_id, player_config, cookies)
                                              File "/home/osmc/.kodi/addons/plugin.video.youtube/resources/lib/youtube_plugin/youtube/helper/video_info.py", line 403, in load_stream_infos
                                                return self._method_get_video_info(video_id, player_config, cookies)
                                              File "/home/osmc/.kodi/addons/plugin.video.youtube/resources/lib/youtube_plugin/youtube/helper/video_info.py", line 738, in _method_get_video_info
                                                parse_to_stream_list(url_encoded_fmt_stream_map)
                                              File "/home/osmc/.kodi/addons/plugin.video.youtube/resources/lib/youtube_plugin/youtube/helper/video_info.py", line 700, in parse_to_stream_list
                                                url += '&signature=%s' % cipher.get_signature(stream_map['s'])
                                              File "/home/osmc/.kodi/addons/plugin.video.youtube/resources/lib/youtube_plugin/youtube/helper/signature/cipher.py", line 24, in get_signature
                                                json_script = function_cache.get(FunctionCache.ONE_DAY, self._load_json_script, self._java_script_url)
                                              File "/home/osmc/.kodi/addons/plugin.video.youtube/resources/lib/youtube_plugin/kodion/utils/function_cache.py", line 101, in get
                                                cached_data = partial_func()
                                              File "/home/osmc/.kodi/addons/plugin.video.youtube/resources/lib/youtube_plugin/youtube/helper/signature/cipher.py", line 57, in _load_json_script
                                                return self._load_java_script(java_script)
                                              File "/home/osmc/.kodi/addons/plugin.video.youtube/resources/lib/youtube_plugin/youtube/helper/signature/cipher.py", line 62, in _load_java_script
                                                raise Exception('Signature function not found')
                                            Exception: Signature function not found
                                            -->End of Python script error report<--
16:36:23.576 T:1886384896  NOTICE: CAESinkPi:AddPackets Underrun (delay:0.00 frames:2205)

shivasiddharth, 2018-02-21:
Its an issue with the Kodi Plugin ref https://github.com/jdf76/plugin.video.youtube/issues/285. Can you check the version of the plugin that you have installed ?

nice824524519, 2018-02-21:
thank you

work for me

I want to contact you business with you. Your email? This part is my enail nice@nicetyshop.com

closed 2018-02-21 by nice824524519