shivasiddharth/GassistPi #222: Helping hand for editing wiki

shivasiddharth, 2018-02-21:
As you may have noticed, with more and more features being added, the README is getting ridiculously long. Its gotten to a point where its become easy to miss out n key and critical info. I have done the fundamental work in creating the wiki, i would like some helping hands to extend the wiki.

jabulonboaz, 2018-02-21:
Happy to help any way i can @Sid. Just let me know how.

ifb777, 2018-02-22:
Happy to lend a hand if I can.  Set up the process and I will devote time to it

aandroide, 2018-02-23:
hello Sid, just today I thought it would be appropriate to divide the program into 3 or 4 versions specifying which functions they contain.
For example, as discussed a few days ago about who owns aiy's sheda audio and has problems with playback using mpv as a player, maybe not interested in playing songs on google play music or turn up and down the volume with the voice and maybe it could go back to using the old script that used vlc as the default media player and would solve the problem.
Even better it would be nice to have a configuration page before cloning your project, where the user chooses what features he would like and based on it he downloads the appropriate script.
what do you think?

shivasiddharth, 2018-02-23:
@aandroide , It would require maintenance of multiple branches and every time a change is made, it would have to be reflected across all the branches which is a nightmare that would consume a lot of time and for that simple reason i would not even attempt that.

closed 2018-05-25 by shivasiddharth