shivasiddharth/GassistPi #223: Installation Failed with exception

amresh1985, 2018-02-21:
I tried twice, but installation failed with exception as 

Setting up ffmpeg (7:3.2.10-1~deb9u1+rpt1) ...
Setting up mpv (0.23.0-2+deb9u2) ...
Processing triggers for libc-bin (2.24-11+deb9u1) ...
Processing triggers for libvlc-bin:armhf (2.2.6-1~deb9u1+rpi1) ...
Requirement already up-to-date: pip in ./env/lib/python3.5/site-packages
Exception:
Traceback (most recent call last):
  File "/home/pi/env/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 594, in urlopen
    chunked=chunked)
  File "/home/pi/env/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 391, in _make_request
    six.raise_from(e, None)
  File "<string>", line 2, in raise_from
  File "/home/pi/env/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 387, in _make_request
    httplib_response = conn.getresponse()
  File "/usr/lib/python3.5/http/client.py", line 1198, in getresponse
    response.begin()
  File "/usr/lib/python3.5/http/client.py", line 297, in begin
    version, status, reason = self._read_status()
  File "/usr/lib/python3.5/http/client.py", line 266, in _read_status
    raise RemoteDisconnected("Remote end closed connection without"
http.client.RemoteDisconnected: Remote end closed connection without response

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/home/pi/env/lib/python3.5/site-packages/pip/commands/install.py", line 353, in run
    wb.build(autobuilding=True)
  File "/home/pi/env/lib/python3.5/site-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/home/pi/env/lib/python3.5/site-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/home/pi/env/lib/python3.5/site-packages/pip/req/req_set.py", line 487, in _prepare_file
    req_to_install, finder)
  File "/home/pi/env/lib/python3.5/site-packages/pip/req/req_set.py", line 428, in _check_skip_installed
    req_to_install, upgrade_allowed)
  File "/home/pi/env/lib/python3.5/site-packages/pip/index.py", line 465, in find_requirement
    all_candidates = self.find_all_candidates(req.name)
  File "/home/pi/env/lib/python3.5/site-packages/pip/index.py", line 423, in find_all_candidates
    for page in self._get_pages(url_locations, project_name):
  File "/home/pi/env/lib/python3.5/site-packages/pip/index.py", line 568, in _get_pages
    page = self._get_page(location)
  File "/home/pi/env/lib/python3.5/site-packages/pip/index.py", line 683, in _get_page
    return HTMLPage.get_page(link, session=self.session)
  File "/home/pi/env/lib/python3.5/site-packages/pip/index.py", line 792, in get_page
    "Cache-Control": "max-age=600",
  File "/home/pi/env/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 501, in get
    return self.request('GET', url, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/pip/download.py", line 386, in request
    return super(PipSession, self).request(method, url, *args, **kwargs)
  File "/home/pi/env/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 488, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/pi/env/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 609, in send
    r = adapter.send(request, **kwargs)
  File "/home/pi/env/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/adapter.py", line 47, in send
    resp = super(CacheControlAdapter, self).send(request, **kw)
  File "/home/pi/env/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/adapters.py", line 423, in send
    timeout=timeout
  File "/home/pi/env/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 643, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "/home/pi/env/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/util/retry.py", line 315, in increment
    total -= 1
TypeError: unsupported operand type(s) for -=: 'Retry' and 'int'

shivasiddharth, 2018-02-21:
Duplicate of #182

closed 2018-02-21 by shivasiddharth

amresh1985, 2018-02-21:
Thanks for prompt reply. I tried below command and then started the installation again. Installation completed successfully.
sudo apt autoremove