shivasiddharth/GassistPi #224: snowboy.service not created

hack-r, 2018-02-21:
To be clear, is there anything in the code that should be creating the snowboy service? If so, I think I've found a bug.

If not, I was just not clear on that from the docs. Thanks!

shivasiddharth, 2018-02-21:
I guess you must be here from the Hackster.io forums. Some of those posts are old and things keep evolving. The snowboy service was removed from the project quite some time back. Always go by the documentation in the git which will always be up to date.

closed 2018-02-21 by shivasiddharth

hack-r, 2018-02-22:
@shivasiddharth  Thank you very much. I see Hackster is as of Dec. 2017 before the snowboy change.

Could you kindly tell me if it's still possible to use custom wakewords with Assistant as before?

shivasiddharth, 2018-02-22:
As of now, nope its not possible. I may add it back sometime in the near future.

hack-r, 2018-02-23:
@shivasiddharth Understood. I am going to go back to an idea I had about trying to use cloudspeech in one thread and assistant in another to achieve this. Failing that, cloudspeech with an action to pipe audio to assistant.