shivasiddharth/GassistPi #227: When i m start google assistant service my pi screen get blank ?

imrushi, 2018-02-22:
hey i m install google assistant when i m start google assistant service my raspberry pi screen turn in to white screen in vnc is working properly and i m stop that service after reboot the pi then work properly. please solve screen blank error.

shivasiddharth, 2018-02-23:
The assistant has no dependencies with the display out. If you can provide some images/screenshots it will help give me a better understanding of your issue.

imrushi, 2018-02-24:
here is screen shot before start service and after stop service.

https://photos.app.goo.gl/4qLeAMnvDQik42FQ2

dwh6411, 2018-02-24:
I think this is the gpio issue, which is conflict with the gpio used by the display spi display .

shivasiddharth, 2018-02-24:
@dwh6411 , Yup this a GPIO issue. @imrushi , Please comment the GPIO usages in main.py and actions.py that overlap with your display GPIOs

imrushi, 2018-02-25:
@shivasiddharth thanks now its working