shivasiddharth/GassistPi #231: Failed to fetch some archives - while installing GA

Curious-Nikhil, 2018-02-24:
While installing Google Assistant for **Raspberry Pi 1 B+** I get this error.
**What should i do?**

Please help! :D
Also great work! @shivasiddharth you are **truly an inspiration!**

[image]

shivasiddharth, 2018-02-25:
Netwrok error. Could be a firewall issue, ISP blockage of debian sources or maybe the pi is not even connected to the network.

Curious-Nikhil, 2018-02-26:
@shivasiddharth  So should I just try to **install** Google Assistant, or should i do something else... 
Thanks!

shivasiddharth, 2018-02-26:
You need to sought out the network issue first. I have already given all the possible reasons, you need to test and figure it out.

Curious-Nikhil, 2018-02-26:
ohk many thnx! :D

shivasiddharth, 2018-03-15:
Closing this for inactivity.

closed 2018-03-15 by shivasiddharth