shivasiddharth/GassistPi #233: debug for the google assistant

bimbim2401, 2018-02-25:
Hello, first thanks for your work. i am impress by it.

i am a new raspberry user .

I want to control my magic mirror "https://magicmirror.builders" with your assistant ( hide / show module).
Maybe do you have made it ? or will do it shortly ?

There is a module call MMM-remotecontrol which listen to get request.

I think i have to change your main.py file to add text comparison and then send http request to the remote module. I have some trouble with it and want to debug you assistant in the terminal but don't know how to send all the print of the main.py into the terminal. can you please help me? 
Can i use the geany IDE to debut gassistPi and show the result of the print in the console / terminal ?

thanks in advance

shivasiddharth, 2018-02-26:
You can attach your main.py here and also the log file, which you can find in /tmp directory. In that way, i can see the changes that you have made and the errors that you are getting. If the language is not listed, then i am afraid, there is nothing i can do, its a google's thing. If you can provide some examples of the https module requests, i shall enable it for public use in the next update.

bimbim2401, 2018-02-26:
In order to work, you have to install first the Magic mirror and the the MMM-Remote-Control module : 
https://github.com/Jopyth/MMM-Remote-Control.git
I have attach my main.py in txt and here is the web example i have found to do it
https://medium.com/@aallan/a-magic-mirror-powered-by-aiy-projects-and-the-raspberry-pi-e6a0fea3b4d6

Thanks so much for your help

[main.txt](https://github.com/shivasiddharth/GassistPi/files/1758002/main.txt)

shivasiddharth, 2018-02-26:
The commands in your main.py are in a wrong place. check the original main.py and you can see the pattern that i have used. 
This is not much of work, i will be releasing an update in the next few days, i shall include this in that. If you are willing to wait, then you need not tinker around the codes..

bimbim2401, 2018-02-28:
Ok i see where to put my command (at the end of the main.py), therefore i am ok to see this features in the next update

thanks for all.

shivasiddharth, 2018-03-01:
Do you have the syntax for the brightness command?

bimbim2401, 2018-03-02:
I think something like
requests.get('http://127.0.0.1:8080/remote?action=BRIGHTNESS&value=150')
range of value is 10 to 200

have you install the magic mirror and the remote controle module to make it works together because i don't yet succeed to hide and show module.
however the google assistant detect my command and send it but nothing happens.

closed 2018-03-02 by shivasiddharth

bimbim2401, 2018-03-05:
Thanks for all. Everything works as expected

shivasiddharth, 2018-03-05:
Can you try adding the brightness command and get back to me, I will then add it to the master. I dont have the resources to run and check MM.

bimbim2401, 2018-03-05:
This command is ok :
http://127.0.0.1:8080/remote?action=BRIGHTNESS&value=150
range of value is 10 to 200

commit 6865d326 (2018-03-02): New Features and Fixes (#230)

Fixes #233 Fixes #79 Fixes #196 Fixes #228 Fixes #229 

Feature additions:
1. Create a emulated philips Hue bridge service.  
2. Turn on/off, change colour, brightness of emulated philips hue lights. 
3. Get recipe info pushed to mobile/tablet.  
4. Track kickstarter campaigns by voice.
5. Remote control of Magic Mirror.