shivasiddharth/GassistPi #238: GassistPi/releases

aandroide, 2018-02-27:
hello can you please update to the January 20 version?
before entering mpv in the scripts.
I prefer to use vlc to play youtube audio.
thank you

shivasiddharth, 2018-02-27:
You want me to downgrade the project ? If you want to make changes please feel free to do but its not good to downgrade the entire project for one person's preference.

closed 2018-02-27 by shivasiddharth

aandroide, 2018-02-27:
no sid maybe I have not explained well, I do not intend to go back the whole project but have the last backup of January 20 available among the releases as you had begun to do. I found it very useful to better customize the project on the basis of each user

shivasiddharth, 2018-02-27:
https://github.com/shivasiddharth/GassistPi/tree/02d3fff9af22095096f644efdc4144672a5682a9  Click on the zip and download option