shivasiddharth/GassistPi #240: Error: During Handling of the above exemption, another exception occured

Curious-Nikhil, 2018-02-28:
**I tried installing again, but it didn't work out.**
Can anyone help me over this?

Many thanks! :D 

[image]
[image]

hamed-ta, 2018-03-09:
I have the exactly same problem , did you solved it ?

dw5zuk, 2018-03-14:
same problem

dw5zuk, 2018-03-14:
`Collecting six>=1.9.0 (from gmusicapi>=11.0.0->-r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 6))
  Downloading six-1.11.0-py2.py3-none-any.whl
Collecting validictory!=0.9.2,>=0.8.0 (from gmusicapi>=11.0.0->-r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 6))
  Downloading validictory-1.1.2-py2.py3-none-any.whl
Collecting appdirs>=1.1.0 (from gmusicapi>=11.0.0->-r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 6))
  Downloading appdirs-1.4.3-py2.py3-none-any.whl
Collecting future (from gmusicapi>=11.0.0->-r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 6))
  Downloading https://www.piwheels.org/simple/future/future-0.16.0-py3-none-any.whl (489kB)
    100% |████████████████████████████████| 491kB 131kB/s 
Collecting proboscis>=1.2.5.1 (from gmusicapi>=11.0.0->-r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 6))
  Downloading https://www.piwheels.org/simple/proboscis/proboscis-1.2.6.0-py3-none-any.whl
Collecting protobuf>=3.0.0 (from gmusicapi>=11.0.0->-r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 6))
  Downloading protobuf-3.5.2-py2.py3-none-any.whl (388kB)
    100% |████████████████████████████████| 389kB 121kB/s 
Collecting decorator>=3.3.1 (from gmusicapi>=11.0.0->-r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 6))
Exception:
Traceback (most recent call last):
  File "/home/pi/env/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 594, in urlopen
    chunked=chunked)
  File "/home/pi/env/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 391, in _make_request
    six.raise_from(e, None)
  File "<string>", line 2, in raise_from
  File "/home/pi/env/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 387, in _make_request
    httplib_response = conn.getresponse()
  File "/usr/lib/python3.5/http/client.py", line 1198, in getresponse
    response.begin()
  File "/usr/lib/python3.5/http/client.py", line 297, in begin
    version, status, reason = self._read_status()
  File "/usr/lib/python3.5/http/client.py", line 266, in _read_status
    raise RemoteDisconnected("Remote end closed connection without"
http.client.RemoteDisconnected: Remote end closed connection without response

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/home/pi/env/lib/python3.5/site-packages/pip/commands/install.py", line 353, in run
    wb.build(autobuilding=True)
  File "/home/pi/env/lib/python3.5/site-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/home/pi/env/lib/python3.5/site-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/home/pi/env/lib/python3.5/site-packages/pip/req/req_set.py", line 554, in _prepare_file
    require_hashes
  File "/home/pi/env/lib/python3.5/site-packages/pip/req/req_install.py", line 278, in populate_link
    self.link = finder.find_requirement(self, upgrade)
  File "/home/pi/env/lib/python3.5/site-packages/pip/index.py", line 465, in find_requirement
    all_candidates = self.find_all_candidates(req.name)
  File "/home/pi/env/lib/python3.5/site-packages/pip/index.py", line 423, in find_all_candidates
    for page in self._get_pages(url_locations, project_name):
  File "/home/pi/env/lib/python3.5/site-packages/pip/index.py", line 568, in _get_pages
    page = self._get_page(location)
  File "/home/pi/env/lib/python3.5/site-packages/pip/index.py", line 683, in _get_page
    return HTMLPage.get_page(link, session=self.session)
  File "/home/pi/env/lib/python3.5/site-packages/pip/index.py", line 792, in get_page
    "Cache-Control": "max-age=600",
  File "/home/pi/env/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 501, in get
    return self.request('GET', url, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/pip/download.py", line 386, in request
    return super(PipSession, self).request(method, url, *args, **kwargs)
  File "/home/pi/env/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 488, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/pi/env/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 609, in send
    r = adapter.send(request, **kwargs)
`

shivasiddharth, 2018-03-14:
Duplicate of #182

closed 2018-03-14 by shivasiddharth