shivasiddharth/GassistPi #241: turn onn wireless device  triggers a service restart

corosuske, 2018-02-28:
hi 

i've been playing with this project, and i love it so far, i've set up a esp8266 with the nodeMCU firmware, and it works sortoff ... but .. when i say "trurn on wireless device one" i get a responce "turning on wireless device one" but then the gassist-ok-google service apears to restart 
the esp8266 seems to work fine if i use the web interface

I might be missing something basic, but i'm not sure 

logs attached below 

Feb 28 22:32:57 raspberrypi python[3840]: ON_CONVERSATION_TURN_STARTED
Feb 28 22:32:57 raspberrypi python[3840]: ON_CONVERSATION_TURN_STARTED
Feb 28 22:33:03 raspberrypi python[3840]: ON_END_OF_UTTERANCE
Feb 28 22:33:04 raspberrypi python[3840]: ON_RECOGNIZING_SPEECH_FINISHED:
Feb 28 22:33:04 raspberrypi python[3840]:   {'text': 'turn on wireless device one'}
Feb 28 22:33:04 raspberrypi python[3840]: Turning On Device One
Feb 28 22:33:09 raspberrypi python[3840]: High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
Feb 28 22:33:09 raspberrypi python[3840]: #011version 1.23.8; written and copyright by Michael Hipp and others
Feb 28 22:33:09 raspberrypi python[3840]: #011free software (LGPL) without any warranty but with best wishes
Feb 28 22:33:09 raspberrypi python[3840]: Directory: /tmp/
Feb 28 22:33:09 raspberrypi python[3840]: Playing MPEG stream 1 of 1: say.mp3 ...
Feb 28 22:33:09 raspberrypi python[3840]: MPEG 2.0 L III cbr32 24000 mono
Feb 28 22:33:11 raspberrypi python[3840]: [0:01] Decoding of say.mp3 finished.
Feb 28 22:33:19 raspberrypi python[3911]: device_model_id: saved-model-id
Feb 28 22:33:19 raspberrypi python[3911]: device_id: 31BCC56019A38FD5A3F1523348E9E513
Feb 28 22:33:19 raspberrypi python[3911]: https://embeddedassistant.googleapis.com/v1alpha2/projects/project-id/devices/31BCC56019A38FD5A3F1523348E9E513 403
Feb 28 22:33:19 raspberrypi python[3911]: [3911:3931:ERROR:assistant_nupnp_client.cc(103)] Cloud discovery nupnp request failed: [60] Peer certificate cannot be authenticated with given CA certificates
Feb 28 22:33:19 raspberrypi python[3911]: ON_MUTED_CHANGED:
Feb 28 22:33:19 raspberrypi python[3911]:   {'is_muted': False}
Feb 28 22:33:19 raspberrypi python[3911]: ON_START_FINISHED
Feb 28 22:33:19 raspberrypi python[3911]: [3911:3931:ERROR:assistant_nupnp_client.cc(103)] Cloud discovery nupnp request failed: [60] Peer certificate cannot be authenticated with given CA certificates
Feb 28 22:33:20 raspberrypi python[3911]: [3911:3931:ERROR:assistant_nupnp_client.cc(103)] Cloud discovery nupnp request failed: [60] Peer certificate cannot be authenticated with given CA certificates

shivasiddharth, 2018-02-28:
Can you please double check if you assigned your ip-address in line 65 (https://github.com/shivasiddharth/GassistPi/blob/master/src/actions.py#L65) within quotes.

corosuske, 2018-03-01:
i have 

the line there is 

IP Address of ESP
ip='192.168.1.53'
#Declaration of ESP names
devname=('Device One', 'lights', 'Device 3')
devid=('/Device1', '/Device2', '/Device3')

the ip is correct (i can visit it from a browser on my laptop and from curl on the raspberry pi 

$ curl 192.168.1.53
<!DOCTYPE HTML>
<html>
<br><br>
Device 1: <a href="/Device1=ON">ON</a> and <a href="/Device1=OFF">OFF</a><br>
<br>
Device 2: <a href="/Device2=ON">ON</a> and <a href="/Device2=OFF">OFF</a><br>
<br>
Device 3: <a href="/Device3=ON">ON</a> and <a href="/Device3=OFF">OFF</a><br>
<br>

</html>

shivasiddharth, 2018-03-01:
You will have a log file in the /tmp folder please attach that here.

corosuske, 2018-03-01:
[GassistPi.log](https://github.com/shivasiddharth/GassistPi/files/1770937/GassistPi.log)
thank you so much

shivasiddharth, 2018-03-01:
In the actions.py script, under the ESP function change the line 173 from `rq = requests.head(ip + dev + ctrl)` to `rq=requests.head("https://"+ip + dev + ctrl)`

corosuske, 2018-03-01:
taht seems to have helped .. i did change it to http and not https ... thank you so very much

closed 2018-03-01 by corosuske