shivasiddharth/GassistPi #242: connecting neopixel on the raspberry

bimbim2401, 2018-02-28:
Hi,

can you please put here a clear schematic for how connect the neoxpixel to the raspberry PI 3. i am not sure to understand the neopixel folder and it seems to be for an arduino.

Thanks in advance.

shivasiddharth, 2018-02-28:
Neopixel uses PWM and Rpi's audio also uses PWM. You can directly connect Neopixel to Pi but audio will not work properly, So the best possible way is to go through a microcontroller/arduino and hence i have given only the arduino/wemos schematics. But if you use APA102 Leds then you can connect it to pi directly without an issue.

closed 2018-02-28 by shivasiddharth

bimbim2401, 2018-03-05:
I dont want to use jack audio, i buy an hifiberry card to connect câble speaker. 

So do you think i need an Arduino ? 

Else there is another solution describe Here 

http://elfboimakingstuff.tumblr.com/post/132956410578/raspberry-pi-pwm-rgb-led-strip

What do you think about it ?