shivasiddharth/GassistPi #244: Gassist with Libreelec

sparx266, 2018-03-04:
Hello

This is a neat project.

I'm trying to get the Kodi control working With Libreelec. Can see that the IP address in actions.py and main.py need to be changed, but the file locations for music will also be different, I'm using an external USB drive, all music is stored in a folder on the root of the drive called 'music'.
So tried /var/media/name of drive/music/ But this didn't work.

Any thoughts on how to get this working?

codelock13, 2018-03-08:
Looks you should try '/media/<name of drive>/music' also you should ask this question in libreelec form instead.

codelock13, 2018-03-08:
It is "/media/name of your drive/music"

shivasiddharth, 2018-03-08:
@sparx266 , you need to have music files and video files in seperate directories. A common directory for both will not work.

sparx266, 2018-03-13:
Thanks for the replies both.
Still the same, no music playback possible on Libreelec.
YouTube playback works, gassistpi can also tell me what's playing and what albums are there, but it won't play what it's told to.

Yes, do realise separate folders for music/video/anything else.
But thanks for pointing it out.

Anyone else got gassistpi and Libreelec talking nicely or can offer some hints?

shivasiddharth, 2018-03-14:
Can you attach the Libreelec and GassistPi logs here?

sparx266, 2018-03-14:
[gassistlog.LOG](https://github.com/shivasiddharth/GassistPi/files/1812903/gassistlog.LOG)

Here is the Gassispi log, the Libreelec log has several files in a zip file, do you need all of it or just some parts?
From the Gassistpi log, it looks like a connection can't be made to the Kodi/Libreelec, yet it found the album and Libreelec is visible on the network.

shivasiddharth, 2018-03-14:
Can you manually run the main.py, give your command and then attach the contents of the terminal as a file here. I have inserted print commands in-between exactly for this reason to identify the source of error.

sparx266, 2018-03-14:
Like this?

(Only 2 CD's on this stick)

[manualoutput.TXT](https://github.com/shivasiddharth/GassistPi/files/1813098/manualoutput.TXT)

shivasiddharth, 2018-03-28:
Sorry for getting back late, had missed the issue. Next time if i don't get back in 4-5 days please feel free to ping me in gitter.

The manual run has printed the tracks and has crashed when requested to print the audio files. The error code denotes an improper parameter which in this case is the directory location/name. Can you please double check the path that you have assigned and make sure that its within quotes.

sparx266, 2018-03-28:
Many thanks for the reply. But still not playing, can confirm that the path is in quotes and is this: "/media/MULTIBOOT/music/"

What is strange is that the album is found, but won't play and crashes the program.  
Any other things to look at?

shivasiddharth, 2018-03-28:
Can you provide me with some more info ? Are you running the libreelec and GA on the same Pi? Also, from the looks of the path, guess the files are on a thumbdrive, has the drive been added to the libreelec media directories?

sparx266, 2018-03-29:
Certainly and many thanks for your help on this. Libreelec and GA are on different Pi's.
Yes the files are on a thumbdrive, each album in a subfolder under 'music'. Kodi can see both albums and displays album art on an attached monitor and a phone app. Kodi can play both albums from keyboard control and the phone app, so have to assume that they are in the media directories. Am new to both Pi and Libbreelec/Kodi so there may be something I've missed?

sparx266, 2018-03-29:
Hang on, re-added the media source and now have something playing, but can't investigate further at the moment. Will report more when time allows.

Thank you once again.

sparx266, 2018-03-30:
OK. Looks like the source must be added a certain way. Through the Files section then scanned into the library. 
Now GA can find and play 'some' files but not all, if it fails it quietly exits the program. Looking in the log it says main error was 'files'.
Is there a maximum number of files as its telling me there are 131 yet there are just over 3000 mp3's on the thumbdrive?

shivasiddharth, 2018-03-31:
Just like in the previous instance, can you pls attach the manual output and the assistant logs?

shivasiddharth, 2018-05-24:
I am extremely for not helping out on time. Are you still stuck ?

sparx266, 2018-05-24:
Yes I am.  This is a huge project and must take up lots of your time.

Any idea what is going wrong here?

Thanks again.

shivasiddharth, 2018-05-24:
The logs and the terminal output seem to contain multiple attempts/reboots, so i am unable to isolate the source of error. If possible, can you please do this:
1. Stop the service.
2. Manually start the service.
3. Give a command that you know will fail.
4. Copy the log file and terminal output onto a separate file and copy it onto some other location.
5. Label them as "Not working GassistPi.log" and "Not Working terminal out.txt"

Repeat a similar process again, but give a command that you know works. This time store the files as "Working GassistPi.log" and "Working terminal out.txt".

Thank you for patience and sticking around. And apologies once again for delayed response.

sparx266, 2018-05-28:
[GassistPi log-not working.txt](https://github.com/shivasiddharth/GassistPi/files/2045809/GassistPi.log-not.working.txt)
[Terminal Output not working.txt](https://github.com/shivasiddharth/GassistPi/files/2045810/Terminal.Output.not.working.txt)
[GassistPi-workinglog.txt](https://github.com/shivasiddharth/GassistPi/files/2045812/GassistPi-workinglog.txt)
[Terminal output working.txt](https://github.com/shivasiddharth/GassistPi/files/2045813/Terminal.output.working.txt)

Thank you.

Here are attached the output files, even though a track is found and played the program still exits.

Your assistance is greatly appreciated, thanks again.

shivasiddharth, 2018-05-28:
Thanks for the files and patience. Will revert back ASAP

shivasiddharth, 2018-06-04:
Hi,
I tried installing kodi and everything seems to work properly without crashing and i was unable to replicate your issue. I used OSMC and native KODI and both worked fine. I can say for sure that the scripts are in order. Though there is not much of a difference between OSMC, KODI and Libreelec, I am not sure if its an issue with Libreelec.

shivasiddharth, 2018-10-07:
Closing this for inactivity.

closed 2018-10-07 by shivasiddharth