shivasiddharth/GassistPi #250: Cannot Unlink Created Device

adas4190, 2018-03-05:
Hi, due to that I made few GassisPi projects in my google home app I have 3 additional devices. Previously I was able to remove them by unlinking, but now it doesn't work. Could anyone please tell me how can I do that or manually remove it pls?

Thank You.

shivasiddharth, 2018-03-05:
It has to be done programatically. Move into the environment and run:
`googlesamples-assistant-devicetool list --device`
This will give you the list of devices registered and their device instance ids. Then run:
`googlesamples-assistant-devicetool delete --device 'Enter your device instance id' `

You need to do it one by one. While testing, i ended up with 100s of devices so for the sake of others having multiple devices, will add a `unlinker script` in the next update.

adas4190, 2018-03-05:
wow, that was quick! Thank You so much... so I didn't have to reinstall every thing all over again :)

Thank You.

adas4190, 2018-03-05:
Hi. What if I change project coz what I did and the only thing that command show me is my current device.

shivasiddharth, 2018-03-05:
If you change the project, the tool cannot be used to delete. I would suggest you to open a issue with the SDK folks regarding the inability to unlink in the app.

adas4190, 2018-03-12:
I have solve it. You have to go google action console website. From there You need add project, from dropdown list pick project You created previously to get json credentials. After You create project on Your right You'll have connected properties,in there you'll have Device models option and From there You can remove registered devices 😊

shivasiddharth, 2018-03-12:
i dont see any connected properties option. can you add a screenshot? i shall  add the instruction in the readme then.

adas4190, 2018-03-12:
I'm sorry but it doesn't work like that. I have got overly excited that's
why I have deleted my post.

adas4190, 2018-03-21:
I belive the issue was fixed by google coz I'm able to unlik all devices I created regardles of credentials.

shivasiddharth, 2018-03-21:
Fair enough closing this thread then !

closed 2018-03-21 by shivasiddharth