shivasiddharth/GassistPi #251: custom command "shut down" not working

Gretehuber, 2018-03-05:
Hi,

thanks for this great work! 
It seems that the custom/local commands like "shut down" do not work. I guess main.py uses actions.py. In there is a check for the key words "shut down" but this doens't seem to work. Any idea how I can track down the problem?

Thanks, Gitti

Rooster69, 2018-03-05:
Try "hey google trigger shutdown" The key word is **trigger**

Wim3d, 2018-03-11:
I have the same issue. Google Assistant works fine, but not the local commands like shutdown. It seems like "trigger" does not trigger on my Pi.

shivasiddharth, 2018-03-15:
Duplicate of #199

closed 2018-03-15 by shivasiddharth

DerakuAng28, 2018-03-16:
This may be silly but I think I figured it out. The code wrote "shut down" with a space in between but Google Assistant parsed it as a single word "shutdown". It works for me after changing the code to "shutdown". Hope this helps.