shivasiddharth/GassistPi #255: using GassistPi with a different user impossible

totomaze, 2018-03-07:
Hi,

I would like to try GassistPi and already change a lot of hardcoded link from /home/pi to home/MyUser, but there are everywhere.

It would be good if it can be set has an environment variable like GASSISTPI_HOME=/home/pi

Great work by the way :)

shivasiddharth, 2018-03-07:
Work is in progress to extend this project to other SBCs and other OSes. Once that commit is merged, the username will not be hardcoded.

k15866, 2018-06-20:
Hey,

Can I just check that it is currently impossible to work this on different users, even if I go into individual files and change all /home/pi to /home/MyUser? I have only read this after troubleshooting multiple issues on installation. I've hit a wall when trying to manually start the assistant as per readme instructions.

I get a traceback which eventuates in:
ImportError: No module named 'yaml'

Despite changing the .py files in /GassistPi/src to look for config.yaml to include MyUser directory, I still come up with the same error. Is there more I need to change, or is there simply no work around and I need to install on /pi directory.

I'm still learning so please be kind :) Thanks

PR #548 (2018-09-28): Remove hardcoded paths
Previously users had to set their usernames to "pi" as all paths were hardcoded. With this PR, the hardcoded paths have been removed and ppl can install the program under any username. Fixes #255
closed 2018-09-28 by shivasiddharth