shivasiddharth/GassistPi #258: Play Music problem

MakeMeCool, 2018-03-11:
When ever I give the command "Hey Google, Play all the songs from Google Music"

It says "Playing All your song"

And after that it says "no matching result found"

I can't able to play music from that commands. Please Help!!

MakeMeCool, 2018-03-11:
but youtube streaming works but play music not woking

shivasiddharth, 2018-03-12:
You need to have music uploaded in the Google music cloud library, it will not play songs that you have on your phone. If you are getting this even after having music in cloud upload the log file here, i shall take a look. The log file will be in /tmp directory

MakeMeCool, 2018-03-13:
here is the log [https://hastebin.com/ohisuvefup.sql](https://hastebin.com/ohisuvefup.sql)

And one more question how to increase the volume.....The Assistant already have the maximum volume but the Output sound of YouTube and Radio is low

MakeMeCool, 2018-03-13:
even I have uploaded songs in my play music cloud

[image]

shivasiddharth, 2018-03-13:
Your log file says that you dont have music in the library. If you added the music files later then use the following command to sync the library and pi songs list:
`"Hey Google, refresh music list"`
For controlling music volume, instructions are given in the readme already : https://github.com/shivasiddharth/GassistPi#ccontrolling-media-by-voice

MakeMeCool, 2018-03-13:
thank you bro for this valuable help!! :)

Lets see....what Happens

MakeMeCool, 2018-03-13:
I refresh the music list but when ever I give any command to play music from google music assistant reboot  I also attaching the screenshot and also the log.
Log >> [https://hastebin.com/eboduyusuh.js](https://hastebin.com/eboduyusuh.js)

Screenshoot >>
[image]

shivasiddharth, 2018-03-13:
Have you added the Google Login credentials in the action.py script? The details are there in readme.

MakeMeCool, 2018-03-13:
yup already there and i give all the credentials (api , email, pass) and also the two steps authentication in my a/c is turned off....

Here is my credentials screenshoot

[https://ibb.co/hvCoCx](https://ibb.co/hvCoCx)

shivasiddharth, 2018-03-13:
Clearly, something is blocking Google API authentication. Are you behind a firewall or proxy ?

MakeMeCool, 2018-03-13:
no

shivasiddharth, 2018-03-13:
I am afraid there is nothing much I can do about this. This is clealry authentication error as the gmusicapi is refusing to login.

legemulo, 2018-03-13:
When I said Play all songs from Google Music it started playing. Here's what happened under the hood:

Mar 13 14:12:51 raspberrypi python[403]: ON_CONVERSATION_TURN_STARTED
Mar 13 14:12:51 raspberrypi python[403]: ON_CONVERSATION_TURN_STARTED
Mar 13 14:12:55 raspberrypi python[403]: ON_END_OF_UTTERANCE
Mar 13 14:12:55 raspberrypi python[403]: ON_RECOGNIZING_SPEECH_FINISHED:
Mar 13 14:12:55 raspberrypi python[403]:   {'text': 'are you all the songs from Google Music'}
Mar 13 14:12:56 raspberrypi python[403]: Playing all your songs
Mar 13 14:12:57 raspberrypi python[403]: High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
Mar 13 14:12:57 raspberrypi python[403]: #011version 1.23.8; written and copyright by Michael Hipp and others
Mar 13 14:12:57 raspberrypi python[403]: #011free software (LGPL) without any warranty but with best wishes
Mar 13 14:12:57 raspberrypi python[403]: Directory: /tmp/
Mar 13 14:12:57 raspberrypi python[403]: Playing MPEG stream 1 of 1: say.mp3 ...
Mar 13 14:12:57 raspberrypi python[403]: MPEG 2.0 L III cbr32 24000 mono
Mar 13 14:12:59 raspberrypi python[403]: [0:02] Decoding of say.mp3 finished.
Mar 13 14:13:00 raspberrypi python[403]: 50.000000
Mar 13 14:13:00 raspberrypi python[403]: 'https://r2---sn-uphcg51pa-bg0e.c.doc-0-0-sj.sj.googleusercontent.com/videoplayback?id=8af23c0d12f8a1d0&itag=25&source=skyjam&begin=0&ei=nAaoWtCyHsiA2gHEoYnYDA&o=15579353635301420259&cmbypass=yes&ratebypass=yes&cpn=8X6TP8vqBhyzuWiLuBwZCA&ip=189.40.44.229&ipbits=0&expire=1520961270&sparams=cmbypass,ei,expire,id,initcwndbps,ip,ipbits,itag,mm,mn,ms,mv,o,pcm2cms,pl,ratebypass,source&signature=82A85944EE1CB70941E4DB50B12DF5303EC66204.1955435342BCE92C893F8612ACB35B6EC60C0EF0&key=cms1&initcwndbps=2490&mm=31&mn=sn-uphcg51pa-bg0e&ms=au&mt=1520961058&mv=m&pcm2cms=yes&pl=19'
Mar 13 14:13:00 raspberrypi python[403]: ON_CONVERSATION_TURN_FINISHED:
Mar 13 14:13:00 raspberrypi python[403]:   {'with_follow_on_turn': False}
Mar 13 14:13:00 raspberrypi python[403]: 50.000000
Mar 13 14:13:00 raspberrypi python[403]: 2018/03/13 14:13:00 socat[15603] E connect(6, AF=1 "/tmp/mpvsocket", 16): Connection refused

shivasiddharth, 2018-03-13:
Issue resolved. CLosing this

closed 2018-03-13 by shivasiddharth