shivasiddharth/GassistPi #260: Install failed at the end

LucaAghemo, 2018-03-13:
Hello, and thx for the amazing work you did.
I have a problem in installing on RPI3b. I followed all the steps running the script "gassist-installer-pi3.sh", and providing the credential file and info of a project I was already using on another RPI3 running Google Assistant succesfully.

The installer reached the end, asked for autorization, I paste the link on the browser and paste the code back to the terminal and, at that point, I had the following error:

https://embeddedassistant.googleapis.com/v1alpha2/projects/xxxxxxxxxxxxxxx/devices/F2757FB52FB81307BF3D3668B3C27660 404
Registering....
[7199:7217:ERROR:assistant_nupnp_client.cc(103)] Cloud discovery nupnp request failed: [60] Peer certificate cannot be authenticated with given CA certificates
[7199:7217:ERROR:assistant_nupnp_client.cc(103)] Cloud discovery nupnp request failed: [60] Peer certificate cannot be authenticated with given CA certificates
[7199:7217:ERROR:assistant_nupnp_client.cc(103)] Cloud discovery nupnp request failed: [60] Peer certificate cannot be authenticated with given CA certificates
Traceback (most recent call last):
  File "/home/pi/env/bin/googlesamples-assistant-hotword", line 11, in <module>
    sys.exit(main())
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/library/hotword.py", line 143, in main
    args.device_model_id, assistant.device_id)
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/library/hotword.py", line 99, in register_device
    raise Exception('failed to register device: ' + r.text)
Exception: failed to register device: {
  "error": {
    "code": 403,
    "message": "Could not create the device instance. You don't have permission to get device model xxxxxxxxxxxxxxxxxx . Make sure you registered this device model under your project xxxxxxxxxxxxxxxxxx.",
    "status": "PERMISSION_DENIED"
  }
}

Of course I double checked to have given the right info, on the right project and so on, nevertheless...

Am I doing something wrong?
Thx in advance

shivasiddharth, 2018-03-13:
Seems like the credentials file that you have used and the project id that you have entered do not match.

LucaAghemo, 2018-03-14:
Thank you, it's strange because in my other googlePi it's working and... even stranger, it works here too!
As an update, despite of the error, Gassist is working! I tried running it, don't ask me why, and I had the surprise.
Something to meditate on :-)
Thx a lot, I have some other issues, but I'll open specific topic for.

shivasiddharth, 2018-03-14:
Closing this issue then

closed 2018-03-14 by shivasiddharth