shivasiddharth/GassistPi #261: General question:

jkings16, 2018-03-13:
Dear Shivasiddharth: 

First of all thank you for the work you put in this project, I've learned a lot from it. 
I have 2 questions:
1) I know that from the latest update wouldn't support snowboy hotword detection. is there any way I could try to make it run again? any instructions on how to install it?
2) do you know any method to change the language of google assistant to Chinese (which is not officially supported yet from Google) I found some examples people changing the language, but I have no clue how did they do it.

I appreciate your work, best regards

Jeffrey

shivasiddharth, 2018-03-13:
I will roll out snowboy in the next update or the update after that. As of now, the only way to change the language would be through the app, if it does not support then there is nothing much that can be done. However you can try cloudspeech. If you can share those examples that you came across, i can try to shed some light on it.

jkings16, 2018-03-14:
I totally appreciate your quick answer, I will check cloudspeech too. 
thank you for your hard work on this. I've learned a lot through it.

best regards

Jeffrey Sebastian

shivasiddharth, 2018-03-14:
Closing this then !

closed 2018-03-14 by shivasiddharth