shivasiddharth/GassistPi #263: Radio fails to play

LucaAghemo, 2018-03-14:
Hello, I have a problem with Radio.
Playing any of the radio station, either the one provided in action.py or other stations taken from the suggested site or from other user's post (therefore supposedly working), generate a replay from the assistant ("tuning into radio "x") but then doesn't work and give a "Connection refused" result in status"
What I mean is that with the command "sudo systemctl status gassistpi-ok-google.service" this is the output I read:

   Loaded: loaded (/lib/systemd/system/gassistpi-ok-google.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2018-03-14 11:20:09 UTC; 59s ago
 Main PID: 1186 (python)
   CGroup: /system.slice/gassistpi-ok-google.service
           ├─1186 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id home-assistant-xxxxxxxxxxxxxxxxxxx
           ├─1237 mpv --really-quiet --volume=10 http://www.radiofeeds.co.uk/bbc6music.pls
           └─1250 /usr/bin/python3 /usr/bin/youtube-dl --no-warnings -J --flat-playlist --sub-format ass/srt/best --no-playlist --format (b

Mar 14 11:20:46 GassistPi python[1186]:         version 1.23.8; written and copyright by Michael Hipp and others
Mar 14 11:20:46 GassistPi python[1186]:         free software (LGPL) without any warranty but with best wishes
Mar 14 11:20:46 GassistPi python[1186]: Directory: /tmp/
Mar 14 11:20:46 GassistPi python[1186]: Playing MPEG stream 1 of 1: say.mp3 ...
Mar 14 11:20:46 GassistPi python[1186]: MPEG 2.0 L III cbr32 24000 mono
Mar 14 11:20:48 GassistPi python[1186]: [0:02] Decoding of say.mp3 finished.
Mar 14 11:20:48 GassistPi python[1186]: ON_CONVERSATION_TURN_FINISHED:
Mar 14 11:20:48 GassistPi python[1186]:   {'with_follow_on_turn': False}
Mar 14 11:20:49 GassistPi python[1186]: 10
Mar 14 11:20:49 GassistPi python[1186]: 2018/03/14 11:20:49 socat[1240] E connect(6, AF=1 "/tmp/mpvsocket", 16): Connection refused

Of course I tried running Gassist from CLI, same result and same message, and I tried dozens of stations, even the one other users normally play. The same radio stream addresses wrote in action.py, used in VLC work perfectly.

shivasiddharth, 2018-03-14:
Please attach the log file. You can find that in /tmp directory

LucaAghemo, 2018-03-14:
Here's the log file, sorry, I forgot.
Thank you.
[GassistPi.log](https://github.com/shivasiddharth/GassistPi/files/1810997/GassistPi.log)

shadowscott, 2018-03-14:
@shivasiddharth Having a similar issue, though when I say 'Hey Google, Radio 1" nothing happens. Can I look somewhere to see what Google is hearing/why it's not actually playing any radio stations?

shivasiddharth, 2018-03-14:
@shadowscott , Hope #204 throws some light. @LucaAghemo , Looks like the radio stations that you have added is not within quotes. Can you double check that ?

shadowscott, 2018-03-14:
@shivasiddharth Thanks. Checked this out earlier and when I try and execute main.py I get this:
pi@raspberrypi:~/GassistPi/src $ chmod +x main.py
pi@raspberrypi:~/GassistPi/src $ ./main.py
Traceback (most recent call last):
  File "./main.py", line 17, in <module>
    from kodijson import Kodi, PLAYER_VIDEO
ImportError: No module named kodijson

shivasiddharth, 2018-03-14:
@shadowscott Please go through the Readme, I have mentioned the correct way to manually run the main.py

shadowscott, 2018-03-14:
@shivasiddharth Ok. I'll take a look. I took a look at #204 and tried 'tune into radio 2' which did say "tuning into radio 2" and started to play. However, I cannot get it to play the other station I listed under a different radio name.

shivasiddharth, 2018-03-14:
@shadowscott We are coming back to the same thing what i mentioned in #204 The name of the radio listed in stname field should match the command deciphered by GA else it will not work. radio2 is not same as radio 2.

If you hit an error, the assistant service will restart, pi does not reboot. To prevent restarts for errors, we need to capture the error. Its a humongous task to test and record each and every error that might occur for a given function, its not possible to roll it out for all the functions in one update i have already done that for few other functions and will do so for other functions one at a time.

LucaAghemo, 2018-03-14:
@shivasiddharth I checked, thank you, but the radio stations are between quotes, as a matter of fact, one of them is the same you wrote in the original file, I didn't touch it. 
I'm afraid it's something else.

shadowscott, 2018-03-14:
@shivasiddharth Ok that makes sense. I'm hoping there won't be any reboots going forward but when there are I can check the log to see what happens. As for the stname field, I changed it to Radio 2 but it doesn't play. When I say "Radio 12" which is the BBC one, it works. So it doesn't like the custom radio link I gave it I guess?

shivasiddharth, 2018-03-14:
@LucaAghemo , Its actually playing the  media then, but volume might be low. Go through the readme instructions for setting media volume. 

From your first comment,
```
Loaded: loaded (/lib/systemd/system/gassistpi-ok-google.service; enabled; vendor preset: enabled)
Active: active (running) since Wed 2018-03-14 11:20:09 UTC; 59s ago
Main PID: 1186 (python)
CGroup: /system.slice/gassistpi-ok-google.service
├─1186 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id home-assistant-xxxxxxxxxxxxxxxxxxx
├─1237 mpv --really-quiet --volume=10 http://www.radiofeeds.co.uk/bbc6music.pls
└─1250 /usr/bin/python3 /usr/bin/youtube-dl --no-warnings -J --flat-playlist --sub-format ass/srt/best --no-playlist --format (b
```
i can see that the media volume is set to 10. At that level, you would barely hear anything.

shivasiddharth, 2018-03-14:
@shadowscott , We are going in circles here. You really need to run the main.py manually to see what GA is deciphering from your command. For all i know, Your Radio 2 command could be deciphered as Radio two. That is what i have mentioned in #204

shadowscott, 2018-03-14:
I see what you mean. I'll try that a bit later and see what's happening.

shadowscott, 2018-03-14:
Also, do you have Telegram? Maybe we can chat on there and you can help me
get Alexa working also. I get emails so if you post it and delete
afterwards I'll see it.

LucaAghemo, 2018-03-15:
Thx a lot, you are right, it was playing but at low volume, saying "Increase Radio Volume 50" made the radio jump out of the speaker :-)
Silly me not thinking about that. Sorry for bothering.

shivasiddharth, 2018-03-15:
Its intentional. All media playback is set to start at previously set volume (just like the original smart speakers).
If you dont need that, then add the following under the stop function in actions.py
`os.system("sudo rm /home/pi/.mediavolume.json")`
and under the:
```
#Function to manage mpv start volume
def mpvvolmgr():
```
change the 50 to whatever number you want between 0 to 100.

Since the primary issue has been resolved closing this thread. Feel free to open it again just in case.

closed 2018-03-15 by shivasiddharth

LucaAghemo, 2018-03-15:
Allow me to suggest one thing, if I can. 
If you use the command "sudo systemctl status gassistpi-ok-google.service" on CLI, you can see how GA understood your last request. Just ask what you want and enter the command, and it will give you the last output.