shivasiddharth/GassistPi #264: ERROR:audio_input_processor.cc(748)] Input error

Curious-Nikhil, 2018-03-14:
[image]

Am using USB Mic and 3.5 mm Jack for Speakers

Please help what should I do
- Raspberry Pi 3 V1.2 B+

Curious-Nikhil, 2018-03-14:
plz help

Curious-Nikhil, 2018-03-14:
so far tried these - 
[link 1](https://www.raspberrypi.org/forums/viewtopic.php?t=185724
)
[link 2](https://www.androidauthority.com/build-google-assistant-raspberry-pi-770296/)

dw5zuk, 2018-03-15:
this also happen to my installation when im running already alexa at the same time...

shivasiddharth, 2018-03-15:
Please check your .asoundrc file. It should be same as asound.conf. Raspbian tends to modify the .asoundrc file on boot sometimes.

Curious-Nikhil, 2018-03-15:
Surprisingly, everything else **works**, pi responds to hotword and gives the reply, only it does **not** show a log when in the terminal

I have made changes, such that both the files have the same content, still **not** a fix.

shivasiddharth, 2018-03-15:
Mostly it has to do with the frequency response of your USB Mic. Try changing your inout device.

Curious-Nikhil, 2018-03-15:
Sure, I'll do that. 
Thanks! :D @shivasiddharth

dw5zuk, 2018-03-20:
is it possible @shivasiddharth  that ican use the same capturing device(usb sound card) with the other application running.. 

/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id bony-196005-20180320121045
ALSA lib pcm_dsnoop.c:618:(snd_pcm_dsnoop_open) unable to open slave
device_model_id: bony-196005-20180320121045
device_id: DD0C92D92FC00190949C511AADF1533D

[2595:2717:ERROR:audio_input_processor.cc(748)] Input error
ON_MUTED_CHANGED:
  {'is_muted': False}
ON_START_FINISHED
[2595:2725:ERROR:audio_input_processor.cc(748)] Input error

im plannig to run the google assistant simultaneously with a SVXLINK(echolink) wich is using also the usb sound input with the assistant..
hope you can help me configure the asoundrc and  /asound.conf

dw5zuk, 2018-03-20:
sudo lsof /dev/snd/pcm*
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
COMMAND PID USER   FD   TYPE DEVICE SIZE/OFF NODE NAME
svxlink 631 root  mem    CHR 116,56          8211 /dev/snd/pcmC1D0c
svxlink 631 root    7u   CHR 116,56      0t0 8211 /dev/snd/pcmC1D0c

shivasiddharth, 2018-05-25:
Closing this for inactivity

closed 2018-05-25 by shivasiddharth