shivasiddharth/GassistPi #265: bug stream playlist youtube

aandroide, 2018-03-14:
hello i tried to say strem playlist singer, i got the link and then i was thrown out of the program and i found that in the link missing a part.
or the red vasco playlist has this url: https://www.youtube.com/watch?v=1CydZtP_XlA&list=PLA782C2BCE7D6DD67
while instead of assistant I get this:
https://www.youtube.com/watch?v=PLA782C2BCE7D6DD67
missing this part of url: 1CydZtP_XlA & list =

shivasiddharth, 2018-03-14:
This is not exactly a bug, its just that i have not programmed it to stream playlists.

aandroide, 2018-03-17:
allright! I also noticed that the autoplay function does not work.

Enzoinventor, 2018-04-07:
Hi, what should be added to the current code to play the playlists?

shivasiddharth, 2018-06-04:
Where do we stand with this ?

aandroide, 2018-06-04:
the error is:
INFO:root:Transcript of user request: "stream playlist Vasco Rossi".
INFO:root:End of audio request detected.
INFO:root:Stopping recording.
INFO:root:Transcript of user request: "stream playlist Vasco Rossi".
stream playlist vasco rossi
stopping vlc
Ottenere il link di YouTube
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:02] Decoding of say.mp3 finished.
playlist vasco rossi
WARNING:googleapiclient.discovery_cache:file_cache is unavailable when using oauth2client >= 4.0.0
Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/__init__.py", line 36, in autodetect
    from google.appengine.api import memcache
ImportError: No module named 'google.appengine'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 33, in <module>
    from oauth2client.contrib.locked_file import LockedFile
ImportError: No module named 'oauth2client.contrib.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 37, in <module>
    from oauth2client.locked_file import LockedFile
ImportError: No module named 'oauth2client.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/__init__.py", line 41, in autodetect
    from . import file_cache
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 41, in <module>
    'file_cache is unavailable when using oauth2client >= 4.0.0')
ImportError: file_cache is unavailable when using oauth2client >= 4.0.0
INFO:googleapiclient.discovery:URL being requested: GET https://www.googleapis.com/discovery/v1/apis/youtube/v3/rest
INFO:googleapiclient.discovery:URL being requested: GET https://www.googleapis.com/youtube/v3/search?key=AIzaSyDVJpyJMbUkudMZjitB7k2RLaAYmSUpJ48&alt=json&q=playlist+vasco+rossi&part=id%2Csnippet
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/pushbutton.py", line 794, in <module>
    main()
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/GassistPi/src/pushbutton.py", line 783, in main
    continue_conversation = assistant.assist()
  File "/home/pi/env/lib/python3.5/site-packages/tenacity/__init__.py", line 241, in wrapped_f
    return self.call(f, *args, **kw)
  File "/home/pi/env/lib/python3.5/site-packages/tenacity/__init__.py", line 330, in call
    start_time=start_time)
  File "/home/pi/env/lib/python3.5/site-packages/tenacity/__init__.py", line 279, in iter
    return fut.result()
  File "/usr/lib/python3.5/concurrent/futures/_base.py", line 398, in result
    return self.__get_result()
  File "/usr/lib/python3.5/concurrent/futures/_base.py", line 357, in __get_result
    raise self._exception
  File "/home/pi/env/lib/python3.5/site-packages/tenacity/__init__.py", line 333, in call
    result = fn(*args, **kwargs)
  File "/home/pi/GassistPi/src/pushbutton.py", line 326, in assist
    YouTube_No_Autoplay(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 1030, in YouTube_No_Autoplay
    audiostream,videostream=youtube_stream_link(fullurl)
  File "/home/pi/GassistPi/src/actions.py", line 375, in youtube_stream_link
    video = pafy.new(url)
  File "/home/pi/env/lib/python3.5/site-packages/pafy/pafy.py", line 124, in new
    return Pafy(url, basic, gdata, size, callback, ydl_opts)
  File "/home/pi/env/lib/python3.5/site-packages/pafy/backend_youtube_dl.py", line 29, in __init__
    super(YtdlPafy, self).__init__(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/pafy/backend_shared.py", line 62, in __init__
    self.videoid = extract_video_id(video_url)
  File "/home/pi/env/lib/python3.5/site-packages/pafy/backend_shared.py", line 51, in extract_video_id
    raise ValueError(err % url)
ValueError: Need 11 character video id or the URL of the video. Got https://www.youtube.com/watch?v=PLA782C2BCE7D6DD67

shivasiddharth, 2018-06-04:
Like I already said, I have not programmed to play playlists.

aandroide, 2018-06-04:
having seen the question, I thought I had added that too.
close if you want too

shivasiddharth, 2018-06-04:
If you check the code, you have the part of script, where I remove a part of the URL to return only the video ids. That function returns the full URL and the id alone. Since you requested a playlist, you get a playlist URL and the script removed a part of it giving this URL that you have now.

aandroide, 2018-06-04:
thanks sid I'll examine this