shivasiddharth/GassistPi #268: [Question] Possible to add a custom answer for a custom input

tordnilsen, 2018-03-16:
Hi.
Thanks for the repo.
Is it possible to create a custom response for a custom input?
For example: "hey [triggerword], is the sky blue?" Answer:"For you it is always blue"

shivasiddharth, 2018-03-16:
You can use the say function to convert the required texts to speech. But, I am just curious, why ?

tordnilsen, 2018-03-16:
Thanks.
Why? :) Just for fun. I want the assistant to 'behave' slightly different than standard answers from google. Not so polite in the answers :)

blackwiz4rd, 2018-03-17:
Have a look at how the radio system is managed in actions.py

mtabba, 2018-04-08:
Sid i have added some libraries in actions.py and it says no module found. i wanted to import cv2 in actions.py But as i did so the google assistant stopped

shivasiddharth, 2018-05-25:
Closing this for inactivity

closed 2018-05-25 by shivasiddharth