shivasiddharth/GassistPi #269: Where are the new features from SDK?

valascus, 2018-03-17:
First of all, i don't know if it's a GAssistPI bug or Google's, so... Just asking.

At March 1st, 2018, Google made an update telling "Support for additional languages, including Spanish (Mexico and Spain), Italian, English (India), Korean, and Portuguese (Brazil). This setting can be passed through the Service API or selected from the Google Assistant app."

In fact, I could, once, pick all those new languages, when it updated, without doing anything. Once i've noticed some major updates to GAssistPI, I've started fresh... And those languages never came back. Now I even have "2 Raspberrys" associated to my Google Account, the one I'm using right now and a older version of GAssistPI that i can't unlink.
Did this happen to one of you?

Thank you ^_^

shivasiddharth, 2018-03-17:
Service API is the pushtotalk script, where you can manually add the language. Regarding the inability to unlink using the app, its a bug that Google needs to resolve. However, you can manually unlink using the Pi itself. For more info checkout #250

shivasiddharth, 2018-03-17:
I have reflected the March 1st update onto this project, but I have held back the March 8th/9th update. As of today, the project works only with RPi boards, at the backend work is going on to put this project on any SBC after that implementation, i shall reflect the March 8/9th update.

shivasiddharth, 2018-05-25:
The project has been updated to SDK 0.5.0.

closed 2018-05-25 by shivasiddharth