shivasiddharth/GassistPi #270: Not checking if pip is installed in gassist-installer-pi3.sh

blackwiz4rd, 2018-03-17:
Hi,
I have seen this error is common: https://imgur.com/a/llYrW
It's probably because you haven't told users to install pip before running.
If it's possible you could include installation in the script.
Thanks for your great work!

blackwiz4rd, 2018-03-17:
I also had to do this after a clean install:
export LC_ALL=C.UTF-8
export LANG=C.UTF-8

valascus, 2018-03-17:
Wich distro did you install?

valascus, 2018-03-17:
Which*

blackwiz4rd, 2018-03-18:
Linux raspberrypi 4.9.80-v7+ #1098 SMP Fri Mar 9 19:11:42 GMT 2018 armv7l

I am just assuming that was the problem because without doing that I couldn't go on with the installation.

shivasiddharth, 2018-03-18:
The error is due to network issues that i have already pointed out in #182 It has nothing to do with pip and language.

closed 2018-03-18 by shivasiddharth

valascus, 2018-03-18:
Also, GAssistPI doesn't support the Lite version of Raspbian. You will have lots of problem to solve if you really want to go that way

shivasiddharth, 2018-03-18:
@valascus , Two updates backed the fixed the installer script to accomodate Lite versions. So there should be no issues.
@blackwiz4rd , The lite versions do not come with pip, that is why i have added pip in the system requirements. So that will be installed.

valascus, 2018-03-18:
Really?!?
A few months ago i tried to use lite and, even with much sacrifice, couldn't make it fully functional. I will test it out then :D