shivasiddharth/GassistPi #271: for the owners of the voice hat google card

aandroide, 2018-03-17:
for the owners of the voice hat google card, you will have noticed that in the gassistpi project a second button is integrated.
in pushbutton.py
pin 22 trigger does not work on this board because it seems to be programmed to be used as a driver.
I solved by setting the gpio 26 that is printed under the servo 0 in the card.
I hope it can be useful to others too.
regards
[https://www.google.it/search?q=gpio+22+voice+hat+google&source=lnms&tbm=isch&sa=X&ved=0ahUKEwjiu8yPhfTZAhUFSBQKHe6_AWkQ_AUICygC&biw=1366&bih=637#imgrc=SdDB0nuZkuJgeM:](url)

kengberg, 2018-04-05:
I'm not sure what you mean by "setting the gpio 26 that is printed under the servo0"?
I see the pin you're talking about - but how do I "set" it?
Any help apriciated..

Enzoinventor, 2018-04-05:
if you have the aiy kit and instead you want to send the commands with the original button and stop the actions with another button, in pushbutton instead of 26 you have to write 23.
And you must also edit the stopbutton.py file from 23 to 26